bigtreetech/smart-filament-detection-module #1: Detection Sensor not working on SKR mini e3 Board

zuverck, 2019-11-21:
Utilized the Marlin build in the SKR mini e3 v1.2 repository to enable the smart filament sensor.  I wired up the sensor to the DET0 (PC12) pin on the board, made the changes to enable the feature in the firmware including printing out the debug info to the terminal.  The firmware never senses a change on the input line of DET0 to reset the runout counter so after 7mm the printer will halt.  I have connected the sensor to a logic analyzer and verified that the sensor is working and toggling the output as filament passes through the rollers.  However their is something wrong between the pin input and the getting into the processor as the PC12 pin never changes state in the software.

Can you provide guidance on your test setup utilized for the skr mini e3 v1.2 and which port was the sensor connected to?

kirelam, 2019-11-27:
Following!

TerawattX, 2019-12-05:
I have run into the same issue.  The default settings in the source recommends PC15, however I'm guessing that's just for on/off switches and not motion.

I created a new Issue on the E3 Mini repo since its likely something with the board and linked back here.

I also saw that Marlin moved from Bugfix-2.0.x to just 2.0.X so I guess that means its released now?  I was going to try compiling straight from there tomorrow and see if that has an effect.

TerawattX, 2019-12-06:
So I just got a response on the other SKR Mini thread stating that PT-DET is not use for filament detection.  I've asked if the module will work on 2 pins vs 3 since technically it comes with a 4 pin to 3 pin cable.  

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/144#issuecomment-562742596

MartinFuersatz, 2019-12-14:
same here - wondering which pin to use on this board

bobkinloch, 2019-12-16:
\marlin-2.0.x\marlin-2.0.x\marlin\src\pins\stm\pins_BTT_SKR_MINI_E3.h
 Change #define FIL_RUNOUT_PIN  PC12   //   PT-DET

#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define NUM_RUNOUT_SENSORS   1     // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.
  #define FIL_RUNOUT_INVERTING true // Set to true to invert the logic of the sensor.                   //***
 // #define FIL_RUNOUT_PULLUP   

M119 to test if the sensor is working
This setup works on my CR10S with the SKR Mini board and a 3 wire sensor

Hope this helps!!

TerawattX, 2019-12-16:
Interesting.  I had the pins set correctly, but I didn't change the FIL_RUNOUT_INVERTING value.  When my current print ends (in a couple of days), I'll update the firmware and see what happens.

Thanks!

TerawattX, 2019-12-18:
@bobkinloch, 

Do you have FIL_RUNOUT_PULLDOWN commented out same as PULLUP?  

I flashed the firmware 2x now to test.  First time I set Inverting to true but missed that you commented out the FIL_RUNOUT_PULLUP as well.  In this state it let me print, but when I removed filament it never seemed to trigger.

I recompiled and left inverting set to true, but commented out the pullup line and now its triggered at all times.  Kind of wish I could change these values in the menu to make testing easier :\

masgar, 2019-12-30:
Anyone solved? I'm facing the same problem. I have also an SKR 1.3 not tried yet. There is a way to check if sensor is faulty?

Thanks

TerryFrench, 2020-01-01:
I just placed the order for one. I will use it with an SKR 1.3 Mini E3 I'm already using. 
I do have an oscilloscope, so I'll likely give it a quick try first. Connect ground and 5V, then check the signal pin between (common ground used for the scope). That way I will see what is expected there. i.e. a high or low signal when there is wire vs. no wire, and also wire moving vs. not moving. 

I'm using 2.0.1 Marlin I configured to use BL Touch. My next step will be to figure out if I want to connect this through Marlin or to a GPIO pin on my Raspberry Pi running Octoprint. If I go the GPIO pin way, I will use some 3v - 5V adapter chip I have running around to not put 5V on the Pi's pins directly. 
Whatever works best... I will post my results here.

phtp, 2020-01-02:
Anyone get this to work with the BIGTREETECH TFT35 V3.0 touchscreen display? They have a Fil-Det pin for this purpose.

CaptainRoot, 2020-01-04:
Had a similar Problem on my Cr10 with MKS SGEN L using Z+ endstop pin. the pin hast teh same pullup,  capacitor wireing as the skr 1.3.

My Problem was that Marlin does not set the runout distance after flashing the firmware.

M412
echo:Filament runout ON
Filament runout distance (mm): 0.00

M412 D7
echo:Filament runout ON
Filament runout distance (mm): 7.00

After that the senor works fine for me

phtp, 2020-01-05:
I have a Ender 3 Pro - I gave up on trying to get it to work with the TFT 35 v3.0 and ended up using the PT-DET on the skr mini e3 v1.2 board.  So far it seems like it does work, but I'll need to do more test around it. 

Not sure if you have a BLTouch as well, but I found the combination of BLTouch and trying to enable the filament runout settings to be over the 256k memory cap.  Did a bit of research and turns out the v1.2 can go above that. 

The modifications I made were on the configs I grabbed are from the bigtreetech github for the skr mini e3 - https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.2/Marlin-2.0.x-SKR-Mini-E3-V1.2  

### platformio.ini - to change from 256k to 512k
[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = **STM32F103RC_bigtree_512K**

My filament runout sensor settings. I trimmed a bunch of stuff and left what's enabled to reduce clutter.
### configuration.h
  #define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define NUM_RUNOUT_SENSORS   1     // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.
  #define FIL_RUNOUT_INVERTING false // Set to true to invert the logic of the sensor.
  #define FIL_RUNOUT_PULLUP          // Use internal pullup for filament runout pins.
  //#define FIL_RUNOUT_PULLDOWN      // Use internal pulldown for filament runout pins.

  #define FILAMENT_RUNOUT_SCRIPT "M600"
  #define FILAMENT_RUNOUT_DISTANCE_MM 7
  #define FILAMENT_MOTION_SENSOR

#define NOZZLE_PARK_FEATURE

### Configuration_adv.h
#define M114_DETAIL

#define ADVANCED_PAUSE_FEATURE

gritstub, 2020-01-13:
@phtp - How is your testing going with the BTT Smart Filament Sensor + SKR Mini E3?  I have the SKR Mini E3 and BLTouch, and am wondering if it's safe to buy the BTT sensor, or if I should look down another path?

phtp, 2020-01-13:
> @phtp - How is your testing going with the BTT Smart Filament Sensor + SKR Mini E3? I have the SKR Mini E3 and BLTouch, and am wondering if it's safe to buy the BTT sensor, or if I should look down another path?

It's pretty much plug and play as long as you adjust the configs right. I just can't get it to work with the touchscreen mode on the tft35.

bakaneko718, 2020-01-14:
@phtp i tried that set up and when i put that in all the sensor does is set off the alarm and the ender 3 doesnt leave boot screen.

phtp, 2020-01-14:
> @phtp i tried that set up and when i put that in all the sensor does is set off the alarm and the ender 3 doesnt leave boot screen.

What does the boot screen say, something along the lines of filament change?

bakaneko718, 2020-01-14:
It just stayed on the marlin splash screen. I went back and took off the 512K. It did the filament change thing with

#define FIL_RUNOUT_INVERTING true

Gave me the full treatment. Maybe about 4 hours into the print it stopped. The filament didn't break and there was no pause screen. It had an ETA message that I hadn't seen before.

phtp, 2020-01-14:
> It just stayed on the marlin splash screen. I went back and took off the 512K. It did the filament change thing with
>
> #define FIL_RUNOUT_INVERTING true
>
> Gave me the full treatment. Maybe about 4 [...]

Do you have v1.0 or v1.2 of the mini board? My settings are for v1.2. Not sure about inverting as mine is set to false. The smart sensor handles run out and jams. Maybe it detected a jam?

bakaneko718, 2020-01-14:
i have v1.2. and it appears that it was jammed. guess its working. now i have to find out why it jams at the extruder.

phtp, 2020-01-15:
> i have v1.2. and it appears that it was jammed. guess its working. now i have to find out why it jams at the extruder.

I had it once where it paused mid-print and I think it was because the filament being too tight. Was a very old PLA roll and near the end of it. I'm not 100% on this, but it hasn't happened with my new roll.

bakaneko718, 2020-01-15:
Seems like the tension on the extruder was too tight. Everything is working now.

TerawattX, 2020-01-16:
I've been tinkering with mine over on [Reddit](https://www.reddit.com/r/BIGTREETECH/comments/ecwbji/btt_smart_filament_detection_module_on_skr_mini/) with the help of (I assume) @phtp.  

@zuverck One thing that was pointed out is that when you call M119, if the sensor is connected it will show "open", but if its unplugged it shows "triggered".  In my case it shows open regardless of whether or not the sensor is attached.  I even ran a jumper between the Signal and Ground wires just to confirm and no change.  Curious if anyone else running into this issue is seeing the same thing.

Robbel2005, 2020-01-22:
> My problem was that Marlin does not set the runout distance after flashing the firmware.
> 
> M412
> echo:Filament runout ON
> Filament runout distance (mm): 0.00
> 
> M412 D7
> echo:Filament runout ON
> Filament runout distance (mm): 7.00
> 
> After that the sensor works fine for me

This solved the same problem on my skr v1.3 too.
If you would  "initialise eprom" after flashing, I'm sure the values will be taken in settings.
But I don't do it every time myself...

btsubjectdata, 2020-01-23:
Is there a light anywhere on this runout sensor? I am having issues as well. It always shows open. SKR E3 Mini 1.2,even have the TFT35 E3 v3 (pins dont work on this for sure, so went with PT-DET PC12).
I am just wondering what is special on the folks who got it to work. I tried invert to true and back to false as well.

btsubjectdata, 2020-01-23:
M119 (not matter if filament or not):
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_min: open
Recv: z_min: TRIGGERED
Recv: filament: TRIGGERED
Recv: ok

btsubjectdata, 2020-01-23:
Also same behavior after this as well:
M412 D7
echo:Filament runout ON
Filament runout distance (mm): 7.00

bakaneko718, 2020-01-23:
ok. so i thought it was working, but i was wrong. sometimes when i start a print it will go to the pause menu and move out of the way as if it was paused. i would resume the print it would be fine. i just did a quick test and cut the filament and let it run, but the sensor didnt kick on.

#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define NUM_RUNOUT_SENSORS   1     // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.
  #define FIL_RUNOUT_INVERTING true // Set to true to invert the logic of the sensor.
  #define FIL_RUNOUT_PULLUP          // Use internal pullup for filament runout pins.
  //#define FIL_RUNOUT_PULLDOWN      // Use internal pulldown for filament runout pins.
  #define FILAMENT_RUNOUT_SCRIPT "M600"
  #define FILAMENT_RUNOUT_DISTANCE_MM 7
  #ifdef FILAMENT_RUNOUT_DISTANCE_MM
       #define FILAMENT_MOTION_SENSOR
  #endif
#endif

CaptainRoot, 2020-01-23:
> Is there a light anywhere on this runout sensor? I am having issues as well. It always shows open. SKR E3 Mini 1.2,even have the TFT35 E3 v3 (pins dont work on this for sure, so went with PT-DET [...]

I wish that there is one. But you can easily add one. I used a cheap red led an a 1k resistor. just solder them together an plug it into the connector of the runout sensor  (between V and S pin). or to have it more permanent just solder it to the connector on the sensor pcb. 
This works just fine for me.

> M119 (not matter if filament or not):
> Recv: Reporting endstop status
> Recv: x_min: open
> Recv: y_min: open
> Recv: z_min: TRIGGERED
> Recv: filament: TRIGGERED
> Recv: ok

this is a motion sensor. so it not just detects is filament is present or not, it toggles between open and triggered when the filament is moving. 
Try to move the filament 1mm then check with M119. do this multiple times  (10 times). if filament is always triggered something is wrong.

An other possibility to check the wiring ist to unplug the sensor an check M119 then shortcut S and V or S and gnd with a 1k resistor and check M119 again, this should definetly toggle the state

btsubjectdata, 2020-01-24:
Thanks all. Tested power and it appears to be getting 5v to the sensor. I ordered a replacement and tried new cable and new smart sensor, same results. 

This is what I have: 

**pins_BTT_SKR_MINI_E3.h**
// Filament Runout Sensor
//
#ifndef FIL_RUNOUT_PIN
//#define FIL_RUNOUT_PIN PC15 // "E0-STOP"
#define FIL_RUNOUT_PIN PC12 // "PT-DET"
#endif

**Configuration.h**
#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define NUM_RUNOUT_SENSORS   1     // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.
  #define FIL_RUNOUT_INVERTING false // Set to true to invert the logic of the sensor.
  #define FIL_RUNOUT_PULLUP          // Use internal pullup for filament runout pins.
  //#define FIL_RUNOUT_PULLDOWN      // Use internal pulldown for filament runout pins.

  // Set one or more commands to execute on filament runout.
  // (After 'M412 H' Marlin will ask the host to handle the process.)
  #define FILAMENT_RUNOUT_SCRIPT "M600"

  // After a runout is detected, continue printing this length of filament
  // before executing the runout script. Useful for a sensor at the end of
  // a feed tube. Requires 4 bytes SRAM per sensor, plus 4 bytes overhead.
  #define FILAMENT_RUNOUT_DISTANCE_MM 7

    #ifdef FILAMENT_RUNOUT_DISTANCE_MM
    // Enable this option to use an encoder disc that toggles the runout pin
    // as the filament moves. (Be sure to set FILAMENT_RUNOUT_DISTANCE_MM
    // large enough to avoid false positives.)
      #define FILAMENT_MOTION_SENSOR
  #endif
#endif

**Configuration_adv.h**
// Show extra position information with 'M114 D'
#define M114_DETAIL

#define ADVANCED_PAUSE_FEATURE
#if ENABLED(ADVANCED_PAUSE_FEATURE)
  #define PAUSE_PARK_RETRACT_FEEDRATE         60  // (mm/s) Initial retract feedrate.
  #define PAUSE_PARK_RETRACT_LENGTH            2  // (mm) Initial retract.
                                                  // This short retract is done immediately, before parking the nozzle.
  #define FILAMENT_CHANGE_UNLOAD_FEEDRATE     10  // (mm/s) Unload filament feedrate. This can be pretty fast.
  #define FILAMENT_CHANGE_UNLOAD_ACCEL        25  // (mm/s^2) Lower acceleration may allow a faster feedrate.
  #define FILAMENT_CHANGE_UNLOAD_LENGTH      100  // (mm) The length of filament for a complete unload.
                                                  //   For Bowden, the full length of the tube and nozzle.
                                                  //   For direct drive, the full length of the nozzle.
                                                  //   Set to 0 for manual unloading.
  #define FILAMENT_CHANGE_SLOW_LOAD_FEEDRATE   6  // (mm/s) Slow move when starting load.
  #define FILAMENT_CHANGE_SLOW_LOAD_LENGTH     0  // (mm) Slow length, to allow time to insert material.
                                                  // 0 to disable start loading and skip to fast load only
  #define FILAMENT_CHANGE_FAST_LOAD_FEEDRATE   6  // (mm/s) Load filament feedrate. This can be pretty fast.
  #define FILAMENT_CHANGE_FAST_LOAD_ACCEL     25  // (mm/s^2) Lower acceleration may allow a faster feedrate.
  #define FILAMENT_CHANGE_FAST_LOAD_LENGTH     0  // (mm) Load length of filament, from extruder gear to nozzle.
                                                  //   For Bowden, the full length of the tube and nozzle.
                                                  //   For direct drive, the full length of the nozzle.
  //#define ADVANCED_PAUSE_CONTINUOUS_PURGE       // Purge continuously up to the purge length until interrupted.
  #define ADVANCED_PAUSE_PURGE_FEEDRATE        3  // (mm/s) Extrude feedrate (after loading). Should be slower than load feedrate.
  #define ADVANCED_PAUSE_PURGE_LENGTH         50  // (mm) Length to extrude after loading.
                                                  //   Set to 0 for manual extrusion.
                                                  //   Filament can be extruded repeatedly from the Filament Change menu
                                                  //   until extrusion is consistent, and to purge old filament.
  #define ADVANCED_PAUSE_RESUME_PRIME          0  // (mm) Extra distance to prime nozzle after returning from park.
  //#define ADVANCED_PAUSE_FANS_PAUSE             // Turn off print-cooling fans while the machine is paused.

                                                  // Filament Unload does a Retract, Delay, and Purge first:
  #define FILAMENT_UNLOAD_RETRACT_LENGTH      13  // (mm) Unload initial retract length.
  #define FILAMENT_UNLOAD_DELAY             5000  // (ms) Delay for the filament to cool after retract.
  #define FILAMENT_UNLOAD_PURGE_LENGTH         8  // (mm) An unretract is done, then this length is purged.

  #define PAUSE_PARK_NOZZLE_TIMEOUT           45  // (seconds) Time limit before the nozzle is turned off for safety.
  #define FILAMENT_CHANGE_ALERT_BEEPS         10  // Number of alert beeps to play when a response is needed.
  #define PAUSE_PARK_NO_STEPPER_TIMEOUT           // Enable for XYZ steppers to stay powered on during filament change.

  //#define PARK_HEAD_ON_PAUSE                    // Park the nozzle during pause and filament change.
  //#define HOME_BEFORE_FILAMENT_CHANGE           // Ensure homing has been completed prior to parking for filament change

  //#define FILAMENT_LOAD_UNLOAD_GCODES           // Add M701/M702 Load/Unload G-codes, plus Load/Unload in the LCD Prepare menu.
  //#define FILAMENT_UNLOAD_ALL_EXTRUDERS         // Allow M702 to unload all extruders above a minimum target temp (as set by M302)
#endif

btsubjectdata, 2020-01-24:
Also, why do they have the pins defined in their user guide in the following format, it appears abnormal. example: "#define FIL_RUNOUT_PIN P1_28"

phtp, 2020-01-24:
> Also, why do they have the pins defined in their user guide in the following format, it appears abnormal. example: "#define FIL_RUNOUT_PIN P1_28"

I was confused about this before, but it's for their other board and not the mini.

btsubjectdata, 2020-01-25:
Does anyone know if the BTT TFT35 E3 3.0 even really supports the Smart Filament Sensor. The document shows "only TFT 3.0" supported. When buying this item, they made it looked like it was supported. 

Incredible how buggy and non functioning this paperweight has proven to be. 

My setup:
BTT Smart Filament Sensor
BTT SKR E3 Mini 1.2
BTT TFT-E3 v3.0

Bigtreetech, feel free to hop in and confirm your products work together with certainty.

adrianmihalko, 2020-01-25:
What's the current situation? Only tft35 connection is not working, but if you connect to motherboard (1.2), it's working?

It's shame that BTT is releasing product without proper support and testing.

btsubjectdata, 2020-01-25:
It doesn’t appear to work no matter where it is connected. Always open (or triggered depending on inverting = true).

I could share the Marlin 2 build of it made sense. But I am really assuming that these are just an issue generally for everyone. 

Does anyone know of a “smart” filament sensor that does distance and clog detection other than the BTT one? Their SKR boards are great, but not 100 % sure about this filament runout sensor haha.

Robbel2005, 2020-01-25:
> It doesn’t appear to work no matter where it is connected. Always open (or triggered depending on inverting = true)

If you do M412 what filament runout distance do you get as answer?

adrianmihalko, 2020-01-25:
Somebody should check this on a scope, in this way we can be sure if filament detector is working properly.

phtp, 2020-01-25:
> It doesn’t appear to work no matter where it is connected. Always open (or triggered depending on inverting = true).
>
> I could share the Marlin 2 build of it made sense. But I am really assuming that [...]

It's a bit of a lengthy thread with TerawattX but maybe you might find this useful. But it might turn out his board is faulty.

https://www.reddit.com/r/BIGTREETECH/comments/ecwbji/btt_smart_filament_detection_module_on_skr_mini/fem2d14/?context=8&depth=9

I posted my configs there and using the BTT github code (https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.2/Marlin-2.0.x-SKR-Mini-E3-V1.2) for the mini versus directly from Marlin.

btsubjectdata, 2020-01-26:
I tried the other firmware and it now says M119 "open" while both connected and disconnected and with and without filament in it.  However, the other firmware I was troubleshooting it said "triggered" in both states.  This is with invert = false

I even tested another "regular" filament runout sensor on my board and it triggers open / triggered respectively.

I have uploaded the firmware I have been working with here:
https://github.com/btsubjectdata/Marlin-SKR-E3-Mini-1.2-bugfix-2.0.x-BTT-SmartFS-BLTouch

If you were to do a compare file between the BTT SKR E3 1.2 vanilla firmware and this you would see a lot of differences, but nothing that I could imagine would stop the sensor from switching. 

Maybe the (potential) bug in this Marlin firmware is the same bug others have reported the same issue as I have. So I am uploading it in the event BigTreeTech would like to figure this issue as it doesn't appear unique.

btsubjectdata, 2020-01-27:
If this code I posted to GitHub works for anyone else using the BTT Smart Filament Sensor, let me know. It would be interesting to see if issue is related to Firware or is hardware related.

TerawattX, 2020-01-27:
@btsubjectdata Sounds like you're in the same boat I am.  I changed my firmware to just a on/off and left it on the PT-DET port to test if I could change the open/triggered state and no matter what it stayed the same.  I'm not sure if this is a fault with the board or bug in the code though.

There is a setting near the end of the configuration_adv file that enables debugging for pins.  Might be worth enabling to confirm if the board is seeing the state change and its just being lost in the code somewhere after that.  By the time I had gotten to this point though I was burnt out and just wanted to print something vs figure out how to even see the pin states, let alone debug more of the code.

In the end I just set up a dumb filament switch using my old Z-stop and wired it to the E0-stop port.  It won't catch clogs and jams, but better than nothing.

Fuyingbro, 2020-03-07:
> I have a Ender 3 Pro - I gave up on trying to get it to work with the TFT 35 v3.0 and ended up using the PT-DET on the skr mini e3 v1.2 board. So far it seems like it does work, but I'll need to do more test around it.
> 
> Not sure if you have a BLTouch as well, but I found the combination of BLTouch and trying to enable the filament runout settings to be over the 256k memory cap. Did a bit of research and turns out the v1.2 can go above that.
> 
> The modifications I made were on the configs I grabbed are from the bigtreetech github for the skr mini e3 - https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.2/Marlin-2.0.x-SKR-Mini-E3-V1.2
> 
> ### platformio.ini - change from 256k to 512k
> [platformio]
> src_dir = Marlin
> boards_dir = buildroot/share/PlatformIO/boards
> default_envs = **STM32F103RC_bigtree_512K**
> 
> My filament runout sensor settings. I trimmed a bunch of stuff and left what's enabled to reduce clutter.
> 
> ### configuration.h
> #define FILAMENT_RUNOUT_SENSOR
> #if ENABLED(FILAMENT_RUNOUT_SENSOR)
> #define NUM_RUNOUT_SENSORS 1 // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.
> #define FIL_RUNOUT_INVERTING false // Set to true to invert the logic of the sensor.
> #define FIL_RUNOUT_PULLUP // Use internal pullup for filament runout pins.
> //#define FIL_RUNOUT_PULLDOWN // Use internal pulldown for filament runout pins.
> 
> #define FILAMENT_RUNOUT_SCRIPT "M600"
> #define FILAMENT_RUNOUT_DISTANCE_MM 7
> #define FILAMENT_MOTION_SENSOR
> 
> #define NOZZLE_PARK_FEATURE
> 
> ### Configuration_adv.h
> #define M114_DETAIL
> 
> #define ADVANCED_PAUSE_FEATURE
> 
> ### pins_BBT_SKR_MINI_E3.h
> #ifndef FIL_RUNOUT_PIN
> //#define FIL_RUNOUT_PIN PC15 // "E0-STOP"
> #define FIL_RUNOUT_PIN PC12 // "PT-DET"
> #endif

Thank you for posting this as I could finally get it working "kind of". I too could not for the life of me get it to work with my tft35 v3.0. So went by your details and put it on my skr e3 mini 1.2.  In 12864 mode the runout portion works, but it keeps thinking the line has stopped. I have tried having the filament going in either end to see if the sensor wheels have to turn one way or the other and does the same no matter what. Oh, and is there a way to have a continue option instead of having to turn off the runout sensor in order to continue. I have either PURGE or RUNOUT SENSOR on/off. Only after I turn the sensor off I get a continue option. For now I am going back to my old runout sensor as I have been at this for days. But any help would be appreciated.

brettvitaz, 2020-03-10:
To anyone still having trouble with this, this comment describes the problem and workaround to the problem: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/144#issuecomment-590925789

The gist is that the PT-DET pin is already claimed by another process. An easy workaround is to initialize the runout sensor after the serial communications.

Essentially, move these lines in `Marlin/src/MarlinCore.cpp`:

```
  #if HAS_FILAMENT_SENSOR
    runout.setup();
  #endif
```

to directly after lines:

```
  #if NUM_SERIAL > 0
    MYSERIAL0.begin(BAUDRATE);
    uint32_t serial_connect_timeout = millis() + 1000UL;
    while (!MYSERIAL0 && PENDING(millis(), serial_connect_timeout)) { /*nada*/ }
    #if NUM_SERIAL > 1
      MYSERIAL1.begin(BAUDRATE);
      serial_connect_timeout = millis() + 1000UL;
      while (!MYSERIAL1 && PENDING(millis(), serial_connect_timeout)) { /*nada*/ }
    #endif
  #endif

  SERIAL_ECHOLNPGM("start");
  SERIAL_ECHO_START();
```

The directions in the manual are otherwise correct (except that the pin is PC12).

SPACEMONKEY-01, 2020-03-11:
Been working this thing for a few weeks now. I do not have a BL Touch. All it took was to hook up the filament sensor to the servo pins PA1 and it works perfect. Not sure what is happening with PC-12, but it's working fine with the PA1 pins.

brettvitaz, 2020-03-12:
Many of us do have a BLTOUCH (myself included). I'm glad to hear you didn't have this trouble.

SPACEMONKEY-01, 2020-03-12:
Oh for sure!! I also want to get a BLT so hopefully they get this pin issue fixed. With my original post I just wanted to let everyone know with a BLT that you're not doing anything wrong. This is a pin issue for sure.

brettvitaz, 2020-03-12:
I'm going to open a PR with the Marlin repo with the fix I mentioned, and open a discussion on how to solve the problem properly.

demonkazuya, 2020-04-22:
Hi sorry for bumping this issue again. Seems i have a different issue with  PC12 PIN (PT-DET). Did all the changes mentioned including the change on Marlin/src/MarlinCore.cpp. Every time i pull out the filament the printer shuts down. I need to reinsert/trigger the sensor the turn on the printer. BTW im using vanila marlin 2.0.5.3. I see that the BTT has been updated to the latest version but I am having a hard time compiling it.

DerDingsCS, 2020-05-03:
I’m having the same problem too.
Tried the fix in MarlinCore.cpp and the board.h fix.
Nothing seems to fix it.

nokesy, 2020-05-03:
If anyone has got this working on a 3 pin fil runout sensor, can they tell me wiring from board to sensor, I have tried (as it should be:
SKR MINI E3 V1.2                  Switch
+5V                                      V
GND                                     G
PC12                                    S

I have tried changing the pins file to run on PC14 (probe). I have left the 5v connection to the PC-DET, but the other 2 (Ground and SIgnal) are on Probe connections.
Results are:
filament in, open
removed, triggered
disconnected  triggered.

Found that it is triggering in reverse, so define FIL_RUNOUT_INVERTING true
this has worked, I am going to try in on the PC-DET connections next... which didn't work at all, so for now I am going to use what works and be happy with it!!

Tazman75, 2020-05-27:
Ended up getting this working. This little vid https://www.youtube.com/watch?v=TrDJmYcZ0Ic was helpful for the setting up the firmware.  The PDF in this repo can be used to work out the wiring.

Once wired up you can do is call M119. Your filament sensor may show open or triggered.  Whichever one it shows, you can advance filament through your sensor just a little bit, then run the command again.  If you wired your sensor up correctly, you should see this flip from Triggered, to open and back and forth.

Once you see your sensor going, you should be able to print.  Make sure to do a M502,
M500 though after your firmware update to reset your settings.  If you don't you can check your M503 results and if you see something like ... "M412 S1 D0.00" it didn't take your FILAMENT_RUNOUT_DISTANCE_MM setting which will cause your system to think filament is jammed/etc. So you'll get the issue where the filament just keeps kicking out.

You can explicitly set this with "M412 S1 D7", and then "M500" to save, but the other way is best.  Good luck.

wat3000, 2020-05-30:
anyone do it? I try Find 12 and 15 and doesn't work
Please comment

blueeagle69, 2020-06-09:
> Had a similar Problem on my Cr10 with MKS SGEN L using Z+ endstop pin. the pin has the same pullup, capacitor wiring as the skr 1.3.
> 
> My problem was that Marlin does not set the runout distance after flashing the firmware.
> 
> M412
> echo:Filament runout ON
> Filament runout distance (mm): 0.00
> 
> M412 D7
> echo:Filament runout ON
> Filament runout distance (mm): 7.00
> 
> After that the sensor works fine for me

If the runout distance is set in the firmware and flashed it won't get used until you run an M502 to reload factory defaults. Confirmed it the other night.
I have the sensor working perfectly connected to the X_Max of my SKR 1.3. Although I had to increase the BTT recommended RUNOUT_DISTANCE_MM from 7mm to 25 in order to stop getting false positives. I'm running a bowden setup with the sensor on top of my BMG clone.

brettvitaz, 2020-08-24:
This issue is resolved in Marlin firmware 2.0.6.

I have verified that using PIN 12 now works without further modification of the source code.

GR8YT1, 2020-08-28:
If not mentioned previously, you can uncomment //**#define FILAMENT_RUNOUT_SENSOR_DEBUG** in **RUNOUT.H**.  The debug information on your console will show you "MOTION DETECTED:0" and "REMAININGMM:#.##" while you're running a print.  You just might have to watch very carefully for it as it scrolls by fast.

LoudMurmur, 2020-10-17:
Also had to do M412 D7, I have a skr 1.3 and I'm running a marlin bugfix of 2 weeks ago (604d10d3075f6a0b9b40fba85aeee5cb3acefb68)