bigtreetech/smart-filament-detection-module #4: filament smart sensor triggers filament change all the time

CsEngineeringDK, 2020-04-19:
I have bought and followed the guide in the link. the sensor triggers a filament runout all the time and I have no idea how to fix the thing. I have the skr mini e3 with the sensor connected to PT-DET as in the guide. I have taken the sensor apart and seems like the wheel is slipping/ not gripping properly. really disappointed with this sensor  https://www.reddit.com/r/ender3/comments/e894j7/marlin_20x_guide_for_ender_3_using_skr_mini_e3_v12/

Tazman75, 2020-05-27:
I had the same problem seems there is an issue with using the PT-DET pin.  Looking at the schematic it should work.  The issue almost seems software related, as I can confirm via the M117 and debug commands the sensor triggers fine off the PT-DET pin, but it doesn't work.  There is a larger thread on this in one of the other threads.

Anyway if you rewire your connect, use the pdf for the pins, and then connect it in there it works.

jyriturkia, 2020-06-01:
https://github.com/bigtreetech/smart-filament-detection-module/issues/1#issuecomment-570776528
Solved issue for me.

brettvitaz, 2020-08-02:
In addition to the filament runout distance not set (for me I just had to restore eeprom to activate my firmware settings), I had to set the distance to 25mm instead of 7mm.

rmpel, 2022-01-31:
Hi @brettvitaz , can I ask _why_ you set to 25mm?

Let me ask in a different way; 
I switched from simple switch to the SFS because of random false positives (sometimes never in a 20 hour print, sometimes about 20 times (!!!!) in a 40 minute print. Quite annoying.
The SFS seems to now display the same erratic behavior, so I suspect software issues, instead of hardware.
Would setting the distance longer prevent these false positives?

Or perhaps @bigtreetech can answer this question for me? Love the products, b.t.w. :) I run an SKR 1.4 with TMC2209 UART on an Ender 3 in this case.

rmpel, 2022-02-22:
Update: my SKR 1.4 no longer accepts new firmware so I swapped it for an SKR 2.0 board, downloaded the latest firmware (Feb20), everything works great, except; still false positives.
I tried FIL_RUNOUT_STATE LOW as well as HIGH, I tried 3mm as well as 10mm for FILAMENT_RUNOUT_DISTANCE_MM.
At this point I will probably remove the sensor and go without, because it's totally unreliable.