bigtreetech/smart-filament-detection-module #6: Filament detection sensor always triggers filament change on SKR 1.4 Turbo.

AbeTheRock, 2020-07-03:
The filament detection sensor keeps triggering for a filament change I've tried inverting the sensor, raised the run out distance but it still happens. I'm using an Skr 1.4 Turbo board and it's plugged into pin slot P1_26. Not really sure how to go about this, any help would be appreciated.

#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define NUM_RUNOUT_SENSORS   1     // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.
  #define FIL_RUNOUT_INVERTING false  // Set to true to invert the logic of the sensor.
  #define FIL_RUNOUT_PULLUP          // Use internal pullup for filament runout pins.
  //#define FIL_RUNOUT_PULLDOWN      // Use internal pulldown for filament runout pins.
  #define FIL_RUNOUT_PIN 2 // Creality CR10-S stock sensor

  // Set one or more commands to execute on filament runout.
  // (After 'M412 H' Marlin will ask the host to handle the process.)
  #define FILAMENT_RUNOUT_SCRIPT "M600"

  // After a runout is detected, continue printing this length of filament
  // before executing the runout script. Useful for a sensor at the end of
  // a feed tube. Requires 4 bytes SRAM per sensor, plus 4 bytes overhead.
  #define FILAMENT_RUNOUT_DISTANCE_MM 7

  #ifdef FILAMENT_RUNOUT_DISTANCE_MM
    // Enable this option to use an encoder disc that toggles the runout pin
    // as the filament moves. (Be sure to set FILAMENT_RUNOUT_DISTANCE_MM
    // large enough to avoid false positives.)
    #define FILAMENT_MOTION_SENSOR
  #endif
#endif

B-rad888, 2020-07-05:
Whats with:
 #define FIL_RUNOUT_PIN 2 // Creality CR10-S stock sensor
I don't see that as an option, or necessary for config.
Skr 1.4 has pins defined in pins file:
 #define FIL_RUNOUT_PIN                     P1_26  // E0DET
#define FIL_RUNOUT2_PIN                    P1_25  // E1DET
Connecting the sensor to 1.26 or 1.25 with the settings you provided above should work
If you are getting false positives make sure :
1. the nozzles is not blocked or too close to the bed
2. the extruder is pushing filament correctly (not skipping or coasting over filament making it appear to be jammed)

AbeTheRock, 2020-07-07:
> Whats with:
> #define FIL_RUNOUT_PIN 2 // Creality CR10-S stock sensor
> I don't see that as an option, or necessary for config.
> Skr 1.4 has pins defined in pins file:
> #define FIL_RUNOUT_PIN P1_26 // [...]

I was also wondering why that line was in the config, I had pulled the config file from the marlin bugfix 2.0 branch for my CR-10 s5. I tried commenting it out and it still didn't work. I made sure the nozzle wasn't too close or jammed and the extruder appears to be working just fine.
One thing I did notice is that when I run an M119 command is that sensor appears open before any filament has been moved. Once the filament moves within the sensor it changes to triggered and remains triggered even when I take the filament out of it. I think it gets stuck as triggered. When I invert it then it gets stuck as open.

B-rad888, 2020-07-07:
Are you printing from a TFT? If so are you using touch mode or marlin mode? 
Did you define in config.h 
#define NOZZLE_PARK_FEATURE
Did you define in config.adv
#define ADVANCED_PAUSE_FEATURE
Are the wires in the correct location?

madk71, 2021-04-07:
Hi, I have the same problem. I use a CR10S with skr1.4 turbo 5x TMC2208 UART(always BTT)+ hemera and a standard CR10s LCD.
The wiring is correct, connected in E0DET, Tried disabling everything (Linear advance, s-curve, junction deviation all off) but nothing.
Tried to print at 1 cm high (basically throwing filament to the bed) and still after a while the change filament is triggered.
Any Idea?

kravviec, 2021-04-14:
Hey guys, I have an ordinary on/off endstop plugged into P1_26 (#define FIL_RUNOUT_PIN P1_26) on BTT SKR 1.4 Turbo. 
The LED on the sensor turns on and off when I switch. But M119 always returns Filament: TRIGGERED. 
I can reverse this state in Marlin (#define FIL_RUNOUT_STATE     HIGH ), but then it is always OPEN. Have you any solution to solve that problem?

madk71, 2021-04-15:
> Hey guys, I have an ordinary on/off endstop plugged into P1_26 (#define FIL_RUNOUT_PIN P1_26) on BTT SKR 1.4 Turbo.
> The LED on the sensor turns on and off when I switch it. But M119 always returns Filament: TRIGGERED.
> I can reverse this state in Marlin (#define FIL_RUNOUT_STATE HIGH ), but then it is always OPEN. Have you any solution to solve this problem?
> [Archive.zip](https://github.com/bigtreetech/smart-filament-detection-module/files/6312827/Archive.zip)

Hi, try to comment #define WATCH_ALL_RUNOUT_SENSORS

kravviec, 2021-04-18:
I've tried, but still nothing happens :(

`#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define FIL_RUNOUT_ENABLED_DEFAULT true // Enable the sensor on startup. Override with M412 followed by M500.
  #define NUM_RUNOUT_SENSORS   1          // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.
  
  #define FIL_RUNOUT_STATE     LOW        // Pin state indicating that filament is NOT present.
  #define FIL_RUNOUT_PULLUP               // Use internal pullup for filament runout pins.
  //#define FIL_RUNOUT_PULLDOWN           // Use internal pulldown for filament runout pins.
  #define WATCH_ALL_RUNOUT_SENSORS      // Execute runout script on any triggering sensor, not only for the active extruder.`

makermelissa, 2021-06-08:
I'm having this issue as well. Nothing is working so far even with nothing plugged in the E0DET port, it counts as triggered.

makermelissa, 2021-06-08:
Also, I have TMC2209 steppers and I suspect there may be a conflict between sensorless homing and the endstop triggers as it is ignoring the X and Y endstops (which is desirable).

makermelissa, 2021-06-08:
Ok, that's the issue. I have a TMC2209 on my E0 extruder. Options are:
1. Disable Sensorless homing on the Extruder Stepper, which should solve it for most people who don't want to mess with the firmware.
2. Swap the FIL_RUNOUT_PIN and FIL_RUNOUT2_PIN values and plug it into the E1DET port for people who don't want to mess with their stepper driver.

I went with the second option because I didn't want to modify my stepper.

ProtomakerSprint, 2021-09-10:
I have the same issue of false triggering; indicating the filament is out even when it is not: ( Bigtree Tech SKR 1.4 Turbo board, Marlin 2.09.01 )

If I turn off Runout detection on the LCD 12864 it 3D prints without issues. Of course this will not detect any filament runout.

So I designed and 3D Printed:-

Clever Filament Runout/Jam Dictator
Use an 3D Printed encoder wheel to trigger an Optical Endstop
Filament actually moving may be viewed by the encoder wheel turning or the Optical endstop- LED flashing
..
NOTE::: **BOTH** the Bigtree Tech Smart Filament Detector and this unit return a signal that the filament is OUT even when the filament is present, false triggering.

Because both do the same false triggering I think it may be a Marlin issue and not either unit.

Looking online indicates false triggering is a common issue.

[https://www.thingiverse.com/thing:4961824]

mlauber, 2022-02-01:
> Are you printing from a TFT? If so are you using touch mode or marlin mode? Did you define in config.h #define NOZZLE_PARK_FEATURE Did you define in config.adv #define ADVANCED_PAUSE_FEATURE Are the wires in the correct location?

Is using a TFT Touch Screen still an issue when running the BTT Smart Filament Runout Sensor?

lsgllc, 2022-03-11:
I am having the same problem. I have tried 1. and 3. Changing to a 2208 did not help.

lsgllc, 2022-03-11:
OK. Here's what I've tried: E3P BTT skr v1.4 turbo sensorless homing on (Since previous post indicate it doesn't matter what you do, these sensors simply do not work.

The best thing to do is leave a flamingly bad review on Amazon and  Google. 

1. Everything mentioned above.
2. found an unused pin and assigned to ROS.
3. tried every combination possible for any defined services using the same pin.

I (We) have runout (TRIGGER) of options. Bottomline: It is a complete hoax that does not work for anyone as far as I can tell.

dsvi, 2022-07-10:
I had the same issue.
The problem was value of `FILAMENT_RUNOUT_DISTANCE_MM` was too small. But when you change it in firmware it doesn't have any effect, since was already saved on the flash, and read from there. So script like 

```
; Enable (and reset) filament runout detection and set runout distance to 30mm
M412 S1 D30 
M500 ; save settings to flash
```
Helped me.