bigtreetech/smart-filament-detection-module #9: Always triggers with Creality V4.2.7 board

bbturbo1, 2020-10-08:
I just installed the new creality V4.2.7 board on an ender 5 and I cannot get this sensor to work. I’ve watched every video I can find and changed my firmware correctly (based on what I’ve been told) even messed with switching between low and high multiple times. I’ve swapped the actual pins on the connector, just to see if that could have been it, and nothing. I just get the printer to do one pass of a purge line and then it tells me to change the filament.

PPPBerlin, 2020-10-16:
Kind of same for me. After a few layer it starts the filament runout procedure. After continue the print it starts the same a few layers after.

`/**
 * Filament Runout Sensors
 * Mechanical or opto endstops are used to check for the presence of filament.
 *
 * RAMPS-based boards use SERVO3_PIN for the first runout sensor.
 * For other boards you may need to define FIL_RUNOUT_PIN, FIL_RUNOUT2_PIN, etc.
 */
#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define FIL_RUNOUT_ENABLED_DEFAULT true // Enable the sensor on startup. Override with M412 followed by M500.
  #define NUM_RUNOUT_SENSORS   1          // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.
  #define FIL_RUNOUT_STATE     LOW        // Pin state indicating that filament is NOT present.
  #define FIL_RUNOUT_PULLUP               // Use internal pullup for filament runout pins.
  //#define FIL_RUNOUT_PULLDOWN           // Use internal pulldown for filament runout pins.

  // Set one or more commands to execute on filament runout.
  // (After 'M412 H' Marlin will ask the host to handle the process.)
  #define FILAMENT_RUNOUT_SCRIPT "M600"

  // After a runout is detected, continue printing this length of filament
  // before executing the runout script. Useful for a sensor at the end of
  // a feed tube. Requires 4 bytes SRAM per sensor, plus 4 bytes overhead.
  #define FILAMENT_RUNOUT_DISTANCE_MM 7

  #ifdef FILAMENT_RUNOUT_DISTANCE_MM
    // Enable this option to use an encoder disc that toggles the runout pin
    // as the filament moves. (Be sure to set FILAMENT_RUNOUT_DISTANCE_MM
    // large enough to avoid false positives.)
    #define FILAMENT_MOTION_SENSOR
  #endif
#endif`

rampardes, 2020-12-17:
I have the same problem.

lstepnio, 2021-01-18:
I'm having the same behavior with the 4.2.2 board. 

Currently using the following firmware to help rule out error in my build config.

https://marlin.crc.id.au/firmware/Ender%203%20-%20v4.2.2%20Board%20-%20BLTouch/

I'm observing sensor state changes using M119 when manually feeding filament through the sensor so physical layer should be good. It's configured for 7mm as verified from the console. Same results from SD or OctoPrint and no issues if filament sensor is disabled with Gcode.

lstepnio, 2021-01-19:
Setting M412 D10  got me up and running. The 7, 8 and 9 would false trigger pretty consistently but so far working without issue with 10mm (D10)

ProtomakerSprint, 2021-01-30:
lstepnio commented 12 days ago
Setting M412 D10 got me up and running. As lstepnio says would false trigger pretty consistently but so far working without issue with M412 D10 = 10mm (D10) 

So I will add  M412 D10  to my Start G Code

https://marlinfw.org/docs/gcode/M412.html

Here are some video details of the Issue I have before using the M412 D10  in my Start G Code:-

https://youtu.be/AN19jHV3Mrk

Modern-Memories, 2022-07-06:
For now I set the runout to 100mm