bigtreetech/smart-filament-detection-module #10: Filament runout detection from onboard SD & TFT35 SD/USB

chris-Neale, 2020-12-12:
Is there a way to link this sensor to both the E0-STOP pin on SRK mini E3 V2.0 and the FIL-DET on the TFT35 E3 V3.0? so that the detection is made with printing from the onboard SD and TFT SD/USB?

oldman4U, 2020-12-21:
kisslorand recommended:
Send M75 to Marlin when print begins. Marlin doesn't know that the print is started so no filament runout warnings. It's not a bug.
You can put M75 in your Start-Gcode and you're set. That's what I did.

Hope this helps, please let us know

Thank you

chris-Neale, 2020-12-28:
Hi, Yes putting M75 in the start G-code has worked fine. I have also added M77 to the end G-code just to be sure.

Thanks for your help.

closed 2020-12-28 by chris-Neale