bigtreetech/smart-filament-detection-module #11: Smart filament sensor non workingwhen attached to TFT35 E3 V3.0

killrob, 2021-01-24:
hi, i have connected the smart filament sensor to the TFT35 E3 V3.0 at the fil-det connector on TFT, this is the part in config.ini
`fil_runout:2
fil_runout_inverting:1
fil_noise_threshold:100
fil_runout_distance:4`
i suppressed all lines comments. 

I had done a 5.30 hours print and at 64% the filament ends, what i expect is that the print pause and the LCD tells me that filament was runned out, but no message and no pause from the LCD, with the result that i have a couple of layers missing in the print.

mlauber, 2022-02-01:
bump.  Did you ever figure this out?  I have the same TFT with an SKR 2.0 rev B and can't get the BTT Smart Filament Runout Sensor to work.

killrob, 2022-02-01:
@mlauber i solved attaching the SFS directly to monitor not to motherboard. But you have to change the config.ini of the monitor like this:
fil_runout:2
fil_runout_inverting:1
fil_noise_threshold:100
fil_runout_distance:7

and disable the LIN_ADVANCE in marlin's Configuration_adv.h and enable M114_DETAIL in the same file

nen777w, 2024-08-24:
> hi, i have connected the smart filament sensor to the TFT35 E3 V3.0 at the fil-det connector on TFT

How do you do that?
SFS 2.0 has wire on 4 pins from one side and 2x2 pins from the other (one for the move sensor and one for the switch sensor).
In TFT35 E3 V3.0,  FIL-DET sensor has only 3 pins.