bigtreetech/smart-filament-detection-module #13: Triggers always on long purgelines

octane321, 2021-02-25:
The sensor triggers on long purgelines even if everything is ok with filament feeding and correct configuration

i.e.: 7mm as recommended. If i do a purge line on start-gcode to clean nozzle and there will be feeded 10mm PLA 1,75 the sensor triggers always and interrupt the print for nothing.....even if i raise up to 15mm it triggers sometimes....with absolutely no problem with filament

it is not only for long and/or thick purge lines....even if you print a long circle with arc welded gcode, it always triggers....that should be really fixed...this i annoying!

thegundamaster, 2021-03-21:
I install mine today and had the same problem

patoberli, 2021-12-26:
Did you ever find a solution for this?