bigtreetech/smart-filament-detection-module #19: What Connect type for the 4 pin Sensor end?

HALLAMNET, 2021-08-13:
Can anyone tell me what type the 4 pin connector is on the sensor?

Its not the same as my skr 1.4T mainboard and the cable sent with it is not long enough and I want to make my own.

Thanks!

mydevpeeps, 2021-09-13:
I don't know if this helps but it's identical to what shipped on my creality.

mydevpeeps, 2021-09-13:
Answer found in this article:
https://www.makenprint.uk/3d-printing/3d-printing-reviews/3d-printer-accessory-reviews/btt-smart-filament-sensor-guide/

Answer: JST-XH

cht685, 2022-01-22:
Negative, the JST-XH is the 3 pin that connects to the Board. Not the Sensor...I want to say the Sensor connector is the JST-ZH, but I cannot find it stated anywhere.

loopj, 2022-12-22:
It's a JST-PH 3 pin