bigtreetech/smart-filament-detection-module #23: Bigtreetech Smart Filament Sensor SFS V1.0 Creality CR-6 & Bigtreetech SKR CR6 V1 Board

michaelwa01, 2021-12-10:
Hello
I wanted to install the Bigtreetech Filament Rolout Sensor, I changed and compiled the firmware according to the instructions. I have replaced the sensor with the original, I can also call up the sensor in the terminal, it brings values, the printer now pushes the filament out and in and breaks off with an error. I have the Bigtreetech SKR CR6 board. 
Is the sensor compatible with the board?

Can someone support me

esunayg, 2021-12-10:
just type the terminal: 
m412 s0
M412 D7
m412 s1
m500

michaelwa01, 2021-12-11:
Thanks for the tip, the Senso is now working,
but I have the problem that the sensor switches off and triggers a runout, but the filamet is 
Loos