bigtreetech/smart-filament-detection-module #28: BTT SKR E3 V3 board and Smart Filament Sensor always triggering as soon as a print is started?

gleep52, 2022-08-20:
The instructions say to invert the FIL_RUNOUT_INVERTING = TRUE.  But that's no longer supported and only FIL_RUNOUT_STATE Low/High is supported.  

I've tried both high and low, and whenever I start a print it immediately triggers the filament unload and loading.  Resumes for another 7mm of filament, then parks and unloads filament again saying it's empty.  

I've tried commending out the RIL_RUNOUT_PULLUP option as a few issues here suggested but did not have success there either.  

Since this is a BTT product and my motherboard is a BTT product - is there something wrong with the default pinouts of their own products for compatibility?  

configuration.h file:

`#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define FIL_RUNOUT_ENABLED_DEFAULT true // Enable the sensor on startup. Override with M412 followed by M500.
  #define NUM_RUNOUT_SENSORS   1          // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.
  #define FIL_RUNOUT_STATE     LOW         // Pin state indicating that filament is NOT present.
  #define FIL_RUNOUT_PULLUP               // Use internal pullup for filament runout pins.
  //#define FIL_RUNOUT_PULLDOWN           // Use internal pulldown for filament runout pins.
  //#define WATCH_ALL_RUNOUT_SENSORS      // Execute runout script on any triggering sensor, not only for the active extruder.
                                          // This is automatically enabled for MIXING_EXTRUDERs.

   // Commands to execute on filament runout.
  // With multiple runout sensors use the %c placeholder for the current tool in commands (e.g., "M600 T%c")
  // NOTE: After 'M412 H1' the host handles filament runout and this script does not apply.
  #define FILAMENT_RUNOUT_SCRIPT "M600"

  // After a runout is detected, continue printing this length of filament
  // before executing the runout script. Useful for a sensor at the end of
  // a feed tube. Requires 4 bytes SRAM per sensor, plus 4 bytes overhead.
  #define FILAMENT_RUNOUT_DISTANCE_MM 7

  #ifdef FILAMENT_RUNOUT_DISTANCE_MM 
    // Enable this option to use an encoder disc that toggles the runout pin
    // as the filament moves. (Be sure to set FILAMENT_RUNOUT_DISTANCE_MM
    // large enough to avoid false positives.)
    #define FILAMENT_MOTION_SENSOR
  #endif
#endif`

Anyone have this thing working on a E3 V3 board?

lyonmaster, 2022-09-11:
I have the same issue and still have not been able to fix it.

GasPath, 2022-09-19:
I have only used the screen connection as it would appear that the TFT is made to work with the SFS. I have worked in the TFT config.ini file, and changing settings there do not seem to change anything for its function.  Both the stock config.ini and my included version do nothing:

#### Filament Runout Sensor
# Select the type of filament runout sensor and its default enabled/disabled state.
#   Options: [Normal Disabled: 0, Normal Enabled: 1, Smart Disabled: 2, Smart Enabled: 3]
fil_runout:3

#### Inverted Filament Runout Sensor Logic
# The sensor uses an inverted logic.
# Disable it in case filament runout is triggered by a LOW signal.
# Enable it in case filament runout is triggered by a HIGH signal.
#   Options: [disable: 0, enable: 1]
# original setting: fil_runout_inverted:1 **NOTE:** I added this line to know the original non-working code
fil_runout_inverted:0

#### NC (Normal Close) Filament Runout Sensor
# The sensor is of type NC.
#   Options: [Normal Open: 0, Normal Close: 1]
#original setting fil_runout_nc:1  **NOTE:** I added this line to know the original non-working code
fil_runout_nc:0

#### Filament Runout Noise Threshold
# Pause print when filament runout is detected at least for this time period.
#   Unit: [time in milliseconds]
#   Value range: [min: 10, max: 1800]
fil_runout_noise_threshold:100

#### Smart Filament Runout Detection
# Used in conjunction with an SFS (Smart Filament Sensor) based on an encoder disc that
# toggles runout pin as filament moves.
#   Unit: [distance in mm]
#   Value range: [min: 1, max: 50]
fil_runout_distance:7

burneme, 2022-09-25:
I had exactly the same problem starting a month ago. Very soon after start the message appeared. 
I'm using a BTT TFT70 which has a plug for the smart sensor. I checked the Smart Filament Sensor with scope ( https://forum.duet3d.com/topic/17107/how-to-setup-the-bigtreetech-smart-filament-sensor/37) , the circuit was fine.
Due to the fact that this a direct drive and the smart filament sensor is loosly coupled with the extruder (hanging below the spool), it can happen that the filament is pulled out much when the printhead goes into the corners and it takes some time until new filament is pulled out.
Solution:
I changed 
#define FILAMENT_RUNOUT_DISTANCE_MM 7 
to 
#define FILAMENT_RUNOUT_DISTANCE_MM 50  
and it works now.
I do hope that this helps you.

sewflynwa, 2023-07-31:
Just resolved this issue on my Ender 3 Pro with SKR mini right after installing the smart sensor.  I cleared the eprom in Marlin, re-entered my z offset and re-accomplished a bed level.  1st layer filament squares print worked perfectly.  No idea what that fixed, but just glad it did.