bigtreetech/smart-filament-detection-module #35: Configuration with sv06 with klipper

foocross, 2023-06-09:
When I add the following to the klipper printer.cfg:

[filament_motion_sensor encoder_sensor]
detection_length: 7
#   The minimum length of filament pulled through the sensor to trigger
#   a state change on the switch_pin
#   Default is 7 mm.
extruder: extruder
#   The name of the extruder section this sensor is associated with.
#   This parameter must be provided.
switch_pin: PA5
# changing the switch_pin name according to your motherboard
#pause_on_runout: True
#runout_gcode:
#insert_gcode:
#event_delay:
#pause_delay:

I get the following error:
The value 'filament_sensor' is not valid for SENSOR

I can't tell if it is working.

GTreaper, 2023-09-30:
Could you tell me how you connected it to the main board of the sovol?

> When I add the following to the klipper printer.cfg:
>
> [filament_motion_sensor encoder_sensor] detection_length: 7
>
> # The minimum length of filament pulled through the sensor to trigger
> # a state [...]

Could you tell me how you connected it to the main board of the sovol?

maydayVFR, 2023-10-08:
I got this same error testing the optical side of the sensor only. I had commented out the switch portion just to isolate the correct pin with fewer variables since sv06 plus was a nightmare to track down the board pinout. With it commented out in printer.cfg my misc-macros file containing the print_start, end etc had no reference to the sensor as it was essentially missing from printer.cfg. Also Check your formatting as I know mine likes to get muddled up by spacing errors especially on mobile edits, as well as spelling, brackets etc.