bigtreetech/smart-filament-detection-module #36: SFS V2.0 with SKR Mini E3 V3

Devo842, 2023-08-24:
Trying to set up the SFS V2.0 on an SKR Mini E3 V3 and not sure of the wiring. The manual has the wiring from the SFS going to 2 pins on the E0DET and to 2 pins on the E1DET plugs and the SKR Mini E3 V3 uses the E0STOP plug for the filament sensor. I am not sure what wires need to go to what pins for the SFS V2.0. Any help would be appreciated.

So6Rallye, 2023-08-25:
Use this reference for your motherboard : https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V3.0/Hardware/BTT%20E3%20SKR%20MINI%20V3.0_PIN.pdf

You need to use E0-STOP and PWR-DET 

Cable red and blue go to E0-STOP
Cable green and black go to PWR-DET

red + 5V
blue (switch sensor) as equal to PC15 (^PC15)
green (motion sensor) as equal to PC12 (^PC12)
black ground

No need to re-pin the cable for your motherboard.

Devo842, 2023-08-27:
Thanks for your help I got it wired up and working kind of. The sensor does not stay blue all the time and would trip as soon as I started a print. To get passed this I increased the detection distance and could get it to print, I cut some filament to test the sensor and it worked as intended. However, it would still trip when there was no problem. The status light on the sensor would go from blue to off multiple times during the print and would still trip mid print with no printing problems. I have the sensor mounted about 4 inches above my direct drive extruder. When the printer is just sitting on, I can move put pressure on the sensor and get the light to flicker blue and off.  Any suggestions on how to remedy this would be greatly appreciated.

So6Rallye, 2023-08-27:
same problem here, I will work on it today.

So6Rallye, 2023-08-28:
The status light on the sensor from blue to off multiple times during the print is normal. if you can print without issues your config is ok.

Devo842, 2023-08-29:
What is your detection distance set at? I am still getting false trips. I have tried a couple of different filaments and moved my detection distance to 100mm just to try and get it to not false trip. The change does prevent it from tripping as soon as the print starts but it will still false trip shortly into the print, it does resume the print well, but it will still trip again. I was checking to see if it kept tripping in the same spot, but it does not appear to be the case especially after I change the detection distance. Any help would be greatly appreciated.

So6Rallye, 2023-08-29:
this is my config for BTT E3 RRF motherboard: 

---------------------------------------------------------------------------
[filament_switch_sensor switch_sensor]
switch_pin: ^PC3
pause_on_runout: False
runout_gcode:
  PAUSE # [pause_resume] is required in printer.cfg
  M117 Filament switch runout
insert_gcode:
  M117 Filament switch inserted

[filament_motion_sensor  encoder_sensor]
switch_pin: ^PE0
detection_length: 20
extruder: extruder
pause_on_runout: False
runout_gcode:
  PAUSE # [pause_resume] is required in printer.cfg
  M117 Filament encoder runout
insert_gcode:
  M117 Filament encoder inserted
---------------------------------------------------------------------------
20 was for the test, think i can reduce to 10. I will try tomorrow. 

make sure on **E0-STOP** you have:

blue > ^PC15 left 
empty > middle
red > +5v right

**PWR-DET:**

green > ^PC12 left 
black > middle
empty > right

[image]

you can try too set pin ^PC12 to !PC12. 

Tell me what's happen with all this infos.

Devo842, 2023-08-30:
So are you using Klipper then, I am using Marlin, either way this gives me some ideas, I will give it a try and let you know how it turns out. I might not be able to get to it till the weekend.

maksii, 2023-09-07:
@Devo842 Did you get it to work?
 the switch works fine for me, but the encoder is always in the detect state.

Devo842, 2023-09-07:
I am still working on it. I tried some firmware changes, and I am still struggling. Any suggestions would be greatly appreciated.

maksii, 2023-09-08:
It's strange how it works, but I can't get it to work without false positives on values less than 30. At 50, it's stable. 
This is good, of course, but 50 is not ~3, so it doesn't make much sense to unpause after such a skip.

Anyway, thanks to @So6Rallye for information.

jfml2020, 2023-09-29:
Something to keep in mind is to look at the mines if the wiring has been changed from the factory connector, the blue light remains fixed and will not flash when moving the filament.

These types of sensors are designed to be next to the extruder to prevent excess filament after moving the extruder from giving an error and pausing.... I have a bontech ddx and I put a sensor on the upper part of the box where I have the printer and I put a Teflon tube from the sensor to the DDX direct extruder so the filament is kept at a certain size and the sensor will always measure the movements. I have it at 3mm working.

sewflynwa, 2024-02-29:
Devo842,
Did you ever get the SFS 2 working correctly?

I have the same board with mine and just cannot get it to work.  The sensor is mounted on top of the extruder and indicates correctly, blinking blue when feeding and solid red when no filament, so I am confident it is wired correctly.  My problem is likely firmware related since the printer never responds to the jam or filament loss, just keeps running merrily along, oblivios to lack of filament.

Very interested in your firmware changes.

jfml2020, 2024-02-29:
> Devo842, Did you ever get the SFS 2 working correctly?
>
> I have the same board with mine and just cannot get it to work. The sensor is mounted on top of the extruder and indicates correctly, blinking [...]

If I have it working 100%, the error is that they do not explain how to use it. Module A is fixed in the printer structure and with a tube as the filament to detect movement 2.8mm Filament a to be in tube Ptfe So motion detector detects it.

sewflynwa, 2024-03-01:
Thank you for looking.

I made changes to 2 files in Marlin, Config.h and Pins.  There were no changes to Config_adv.h.  There were other changes not shown here specific to my extruder for loading and unloading distances and feed rates etc., but those should not affect the sensor working/tripping a runout.  

Inputs welcome...

Configuration.h changes
#define FILAMENT_RUNOUT_SENSOR   //Uncommented
#define FILAMENT_RUNOUT_DISTANCE_MM 3  //Uncommented, 3
#define FILAMENT_MOTION_SENSOR  //Uncommented
#define FILAMENT_SWITCH_AND_MOTION  //Uncommented.
#define FIL_MOTION1_PIN   PC12  //Uncommented, changed -1 to PC12 for movement sensor
#define FIL_MOTION1_PULLUP   //Uncommented
#define ADVANCED_PAUSE_FEATURE //already uncommented

Configuration_adv.h changes
#define ADVANCED_PAUSE_FEATURE //already uncommented

Svc/pins/STM32G0/pins_BTT_SKR_Mini_E3_V3_.0h changes
 #define FIL_RUNOUT_PIN       PC12  //changed from PC15.  //E0-STOP
 #define POWER_LOSS_PIN     -1  //changed from PC12, Power Loss Detection: PWR-DET

sewflynwa, 2024-03-02:
SFS 2 works perfectly!!  But not when running the machine from the TFT 35 touchscreen in other than Marlin mode.  In Marlin mode, the SFS motion sensor kicks in, M600 kicks in and unloads the filament, insert new filament prompt, put in new filament, and off it goes.  Happy about it working at least, I have some lengthy prints I was reluctant to start without a full spool of filament.  Now I am going to have to shift gears and start looking at the TFT 35.  There has to be some kind of pin conflict.

Highly suggest trying your SFS 2 from Marlin screen if you are knocking your brains out attempting every change you can think of to try to get it to work.  

A few things I did notice: 
1) The filament sensor settings entered into the Marlin firmware, load/unload length, rates, etc, were not what was reflected in the filament runout sensor settings in the Marlin menu settings. They appeared to be appropriate for a base machine without any upgrades.
2) Shifting to the TFT Screen mode, even without turning the machine off, and starting another print from the ports on the TFT resulted in the SFS not working.  I have not tried initiating a print from the SD card in the Ender from the TFT screen yet, but prints from either of the ports on the TFT resulted in SFS not working.

I am really hoping a firmware update for the TFT is available that will resolve this issue, and will look at it tomorrow.

viyer00, 2024-03-03:
Sewflynwa's solution was the first that has worked for me after many days of trying. I am using an SFS v2 with SKR Mini E3 v3 and TFT35 on Marlin mode, sending prints through octoprint/usb, and with the following setup: filament -> SFS -> dual gear extruder -> bowden -> all metal hotend. Previously, no matter what I did with the firmware, the sensor would trigger at the beginning of every print, even at a 50mm runout distance. Now it does not trigger at the beginning of a test print with a 5mm runout distance, and it properly triggered when I cut the filament as a test. Thanks!

sewflynwa, 2024-03-03:
After several hours reading, compiling, and uploading, this is what I think is going on.  The TFT interface with the skr mini is not as seamless as one would think.  When the file to print is selected from the tft sd or usb port, the skr mini marlin job timer is not started.  This is one of the basic requirements for the filament sensor to work.  No timer, no filament sensor working.  This is also why when the file is selected in Marlin from the skr sd port, the filament sensor works.

There is a commented out instruction at the end of the tft config.ini file that says to enable gcode start, end,  and pause by changing their values from0 to 1.  It then states to enable start timer and end timer, M75/n and M77/n.  Not clear on what this means or where to do this.  So I tried it in the config.ini file, uploaded to the tft, with no change. I then tried it in the pre-print slicer script and the post print slicer script.  This sorta worked yielding unstable filament sensor stops so I deleted it from the scripts.

Thatis as far as my patience allowed me to go right now.  Sent an email to BigTreeTech asking questions.  Will see what happens.

viyer00, 2024-03-03:
I had added the M75/M77 to the beginning/end of the slicer settings before reading your solution, and it more or less seems to work over the last few prints, but I will delete it to see if it is more stable. When I set runout to 3mm last night, the SFS triggered at the very beginning of a 10h print, but I reloaded the filament, reenabled filament runout during the same print, set the runout to 4mm, and successfully completed the print. I am not sure if the false positive was due to my slicer M75/M77 settings, the 3mm distance, or something else. I will test more over the next few days for stability. I also recall that the TFT35 firmware recommends to set the FIL-DET to 0 when the SFS is connected to the motherboard, but I haven't tried that yet. 

Separately, is it possible to use the BTT 24v UPS while also using the SFS v2 that uses the PWR-DET pin? I believe the only alternative pins are E0 stop (also used by the SFS v2) and the neopixel (I plan on using a neopixel strip)--is this correct? I notice that the BTT shutdown relay 1.2 has a PWR-DET header, but I am not sure if I can daisy chain the motherboard -> shutdown relay -> 24v UPS in this way.

sewflynwa, 2024-03-03:
I will try changing the runout distance to 4 as well.  The other option I have not tried is to manually edit the gcode file to put in the M75/M77 rather than using the pre and post script.  I still find it shocking that the board and TFT have this kind of conflict after being advertised as compatable and with no fix after having been out for so long.

Unfortunately, I am not knowledgeable about the UPS.  I did see when making the changes to the pins file for the SFS the power loss was also assigned PC12 just a few lines lower in a power loss section.  I would think any other open pin may work as the board seems to be very flexible in that separtment.

viyer00, 2024-03-06:
The above settings have been working flawlessly for me in Marlin mode on the TFT35 E3 v3 over long prints at 4mm distance. However, I've had some strange problems when switching between touchscreen mode. In the middle of a print today, I switched to touchscreen mode, reenabled the filament sensor (that setting does not automatically carry over from Marlin), and had no problems for over an hour. I switched back to Marlin mode during that same print, and some minutes later the screen had artifacts and froze, apparently triggering some part of the filament sensor and throwing up all sorts of "paused click to continue" on the Marlin screen and in Octoprint. On the 20th try of me pressing continue it finally registered the input and went back to printing. The filament sensor is still enabled, and I think this print will end up fine,

jfml2020, 2024-03-06:
> The above settings have been working flawlessly for me in Marlin mode on the TFT35 E3 v3 over long prints at 4mm distance. However, I've had some strange problems when switching between Marlin and the touchscreen mode on the TFT. In the middle of a print today, I switched to touchscreen mode, reenabled the filament sensor (that setting does not automatically carry over from Marlin), and had no problems for over an hour. I switched back to Marlin mode during that same print, and some minutes later the screen had artifacts and froze, apparently triggering some part of the filament sensor, nozzle park, and eventually throwing up all sorts of "paused click to continue" on the Marlin screen and in Octoprint. On the 20th try of me pressing continue it finally registered the input and went back to printing. The filament sensor is still enabled, and I think this print will end up fine, but staying in Marlin mode definitely seems more stable.

Since it comes out from the sensor to the strusor, is the filament free without a tube? When doing reactions, if the filament comes and goes but the sensor does not, if it does not have a tube, the filament bends when retracting, The sensor does not detect these small movements because the filament twists or bends to the sides, the tube means that the filament is always detected leaving and entering by the movement sensor.

leonardojc, 2024-05-21:
> Thank you for looking.
> 
> I made changes to 3 files in Marlin, Config.h, Config_adv.h, and Pins. There were other changes not shown here specific to my extruder for loading and unloading distances and feed rates etc., but those should not affect the sensor working/tripping a runout.
> 
> Inputs welcome...
> 
> Configuration.h changes #define FILAMENT_RUNOUT_SENSOR //Uncommented #define FILAMENT_RUNOUT_DISTANCE_MM 3 //Uncommented, 3 #define FILAMENT_MOTION_SENSOR //Uncommented #define FILAMENT_SWITCH_AND_MOTION //Uncommented. #define FIL_MOTION1_PIN PC12 //Uncommented, changed -1 to PC12 for movement sensor #define FIL_MOTION1_PULLUP //Uncommented
> 
> Configuration_adv.h changes #define ADVANCED_PAUSE_FEATURE //already uncommented
> 
> Svc/pins/STM32G0/pins_BTT_SKR_Mini_E3_V3_.0h changes #define FIL_RUNOUT_PIN PC12 //changed from PC15. //E0-STOP #define POWER_LOSS_PIN -1 //changed from PC12, Power Loss Detection: PWR-DET

Hello @sewflynwa,  have you been able to solve this? I'm with the same problem

I see that in your definitions you use the same PIN for both sensors. this is correct?
in Configuration.h
**#define FIL_MOTION1_PIN PC12** // changed -1 to PC12 for movement sensor
and in pins_BTT_SKR_Mini_E3_V3_.0h
**#define FIL_RUNOUT_PIN PC12** //E0-STOP

sewflynwa, 2024-05-22:
I have not been able to solve the problem.  The sensor does work correctly when the machine is running in Marlin mode.  This requires use of the mini plug on the front of the machine to load the files which is something I wanted to stop doing with the TFT35.

The pin being assigned to both snsors has not been a problem for me.  Again, if the machine is running in Marlin mode.  When the machine is running in touch mode, the sensors simply do not stop the print when tripped.

The real problem appears to be the way the TFT Board interfaces with the motherboard when the machine is running in touchscreen mode.  Of course, this is the mode required to use the SD and USB ports on the TFT35.  I am hopeful that a later TFT 35 firmware version will fix the issue.  For now, I just use Marlin mode when it appears there may not be enough filament on the spool to complete the print.

Coolbobjoe, 2024-09-12:
I have bought two now, neither will ever turn their blue lights on nor show up on klipper. support has went awol on email

sewflynwa, 2024-09-13:
Unfortunately, not familiar with Klipper and I gave my machine to a youngster 2 weeks ago for him to use in his high school technology class.  I  moved to a Prusa mK4.  However, I do recall I found the SFS wiring to the E3V3 board was very confusing for me.  That, and the changes required for the Marlin firmware yielded many starts with no lights before I got it correct.

BTT support checked out on me as well when the issue started looking more and more like a compatibility problem with the board, TFT 35 touch screen, and the SFS 2.0 when operating in touch mode.  As I mentioned above, marlin mode worked as expected.

jfml2020, 2024-09-13:
Remember that if you do not have Teflon tube between the sensor and the extruder you will have false negative problems or the sensor is placed directly on top of the extruder or the tube is placed between the extruder and the sensor so that filament slides back and forth millimeters on sensor.