bigtreetech/smart-filament-detection-module #40: SKR Pico Connection?

dgpmaluco, 2023-09-21:
Dear BTT,

how would you connect the SFS 2.0 to a SKR Pico?

best regards

jonathanalevi, 2023-09-27:
Add this to the appropriate config file on YOUR setup (on mine it's sky-pico-v1.0.cfg)

```
[filament_switch_sensor switch_sensor]
switch_pin: ^gpio16 # as shown in Figure 2, switch sensor I is PAO
pause_on_runout: False
runout_gcode:
  PAUSE # [pause resume] is required in printer.cfg
  M117 Filament switch runout
insert_gcode:
  M117 Filament switch inserted

[filament_motion_sensor encoder_sensor]
switch_pin: ^gpio20 # as shown in Figure 2, motion sensor IO is PC2 
detection_length: 2.88 # accuracy of motion sensor 2.88mm 
extruder: extruder 
pause_on_runout: False 
runout_gcode:
  PAUSE # [pause resume] is required in printer.cfg
  M117 Filament encoder runout
insert_gcode:
  M117 Filament encoder inserted

```

And here's what it should look like with yellow highlights to point out:
[image]

Someone at BTT should add this as a sample config. I don't know how to submit a pull request, I'm not a programmer.

dgpmaluco, 2023-09-27:
Thanks I will try this!

SirRenix, 2023-10-12:
Unfortunately this will not work with the standard cabling on FAN3 of Voron V0.2 config, because according to schematic this port GPIO20 is linked with the laser port. Furthermore Klipper gives an error message that the port is already occupied. You are not allowed to connect a fan on FAN3 and have to switch to FAN 2/1.

idubrov, 2024-02-16:
>Not sure if GPIO29_ADC being an ADC port will cause an issue or not.

Oh yes, it will. Apparently, GPIO29 is used internally by RP2040 to measure VSYS, and connecting it as a digital input messes up thermistor reads somehow, causing Klipper to crash with "ADC out of range".

However, there is a pin header on a board with GPIO22 on the middle pin -- this one could be used for encoder. I re-pinned the connectors such that one connector, JST, got red/black/blue and is plugged into E0 stop and another connector, Dupont, gets a single wire, green, going to that middle pin with GPIO22.

anthonielamay, 2024-03-19:
Has there been any updates or solutions for any of you?

howdy-vulcan, 2024-05-25:
I got this working for anyone who is looking to connect the BTT smart filament sensor to a SKR Pico.  There doesnt seem to be enough points for a correct separate connection.  I was able to feed a needle though the connection and swap the pins around on the Motion / GND connnection.  I matched the RGB connection layout per the pinout diagram on the Pico (next to the E0 endstop)    and activated it via that pin. Changing the connector pins allows for more ports to be used.

Sentinel-sys, 2024-06-14:
> I got this working for anyone who is looking to connect the BTT smart filament sensor to a SKR Pico. There doesnt seem to be enough points for a correct separate connection. I was able to feed a [...]

Hi there,

could you please go into more details please. I have the same problem here. I used 5 Volt and GPIO16 from the E0-Stop port and GND and GPIO26 from the SERVOS port.
It seems this triggers a restart from the Pico board. Or better, Klipper reports an MCU shutdown.

Thank you

Joe

idubrov, 2024-06-14:
> could you please go into more details please. 

The key moment here is that you cannot use 5V supply for the sensor if you want to use SERVOS (GPIO29) port.

@howdy-vulcan seems to be using RGB port (which was not available in my case).

[I used GPIO22](https://github.com/bigtreetech/smart-filament-detection-module/issues/40#issuecomment-1947788690) instead of GPIO29 (SERVOS), and it works fine since. Look for the [P_S IO22 PROBE](https://github.com/bigtreetech/SKR-Pico/blob/master/Hardware/BTT%20SKR%20Pico%20V1.0-PIN.pdf) connector on the diagram.

Now that I think about it, maybe a simpler solution would be to put switch encoder on GPIO29 (SERVOS) and motion sensor on GPIO16 (E0 stop). If encoder switch button is wired the way buttons are commonly wired (normally connected, signal to ground), there will be no issue with 5V flowing back through the button. And GPIO16 should be able to handle 5V flowing back into it. But need to check if button is actually wired that way.

Sentinel-sys, 2024-06-14:
> Now that I think about it, maybe a simpler solution would be to put switch encoder on GPIO29 (SERVOS) and motion sensor on GPIO16 (E0 stop). If encoder switch button is wired the way buttons are [...]

I'm a bit rusty when it comes to this. But just tell me what I need to measure and I'll do it for you.

Greetings
Joe

Sentinel-sys, 2024-06-14:
> [I used GPIO22](https://github.com/bigtreetech/smart-filament-detection-module/issues/40#issuecomment-1947788690) instead of GPIO29 (SERVOS), and it works fine since. Look for the [P_S IO22 [...]

Do you used GND from the E0-Stop? Because I did this now (GND, 5V, GPIO16 from E0-Stop) and GPIO22 from PROBE. Switch is working Encoder not.

Sentinel-sys, 2024-06-14:
> When you say encoder doesn't work, do you get blinking from it at all (when you move filament through it, it should blink LED in the sensor itself)?

Red light for the switch is working when no filament is present. But no blue light at all.

idubrov, 2024-06-14:
>Switch is working fine. I see the red light when no filament is loaded and in Klipper it shows as "Empty". Encoder light is blue blinking when filament is pushed trough but I don´t know how to test the encoder inside Klipper. It always says "Detected".

It will only detect when you ask it to extrude filament, but no motion is detected. Remove your filament from the extruder (but maybe leave it in the switch so it still shows "detected"), extrude some length, and see if it goes into "not detected" for the motion sensor.

Sentinel-sys, 2024-06-14:
I almost thought so. Only the combination of "extruding" and "filament not moving" triggers the alarm...
Good to know.
Many thanks for your help. I think that works for me so far. 

> P.S. Also, maybe measure a voltage between GND and the switch wire to make sure it is not feeding 5V back.

I measure around 3.5V between GND and the switch pin and around 4.5V between GND and the encoder pin.

idubrov, 2024-06-14:
>I measure around 3.5V between GND and the switch pin and around 4.5V between GND and the encoder pin.

That is still more than 3.3V. I guess, that red LED is what provides that connection from 5V. Well, if you don't see any voltage-induced errors (like temperatures jumping around), then it's fine. Or you can go back to GPIO22 idea. Not sure why didn't it work for you...

Sentinel-sys, 2024-06-14:
Okay thank you. I will finish my build and amke some more tests. Maybe I switch back to your idea.

howdy-vulcan, 2024-06-14:
> > Do you used GND from the E0-Stop? Because I did this now (GND, 5V, GPIO16 from E0-Stop) and GPIO22 from PROBE. Switch is working Encoder not.
> 
> Yes, that's what I did. All GND are connected together on the board. For the reference, this is what my config looks like:
> 
> ```
> [filament_switch_sensor switch_sensor]
> switch_pin: ^gpio16
> pause_on_runout: False
> 
> [filament_motion_sensor encoder_sensor]
> switch_pin: ^gpio22
> extruder: extruder
> ```
> 
> When you say encoder doesn't work, do you get blinking from it at all (when you move filament through it, it should blink LED in the sensor itself)?

So I actually had the same thing happen, the encoder  kept triggering a pause (though had correct red/ blue lights)  so i commented out the encoder from the config, and removed the black / green lines. i had connected to the RGB Port on PIco.   For last few weeks, no issues with this.   I ended up installing some Neopixels Strips  and Servos for Nozzle Wipe + Zeroclick Probe on the Pico with some additional ports from Klipper Expander

IMO  even though i didnt get the encoder working right,  for a V0 - its small and prints so fast that consequences of a tangle or breakage are not terrible in most cases.  

my config  sections for reference with rough debug messages: 

```
[filament_switch_sensor switch_sensor]
switch_pin: ^gpio16
pause_on_runout: False
runout_gcode:
 PAUSE # [pause_resume] is required in printer.cfg
  M117 Filament switch runout
  M118 Filament Switch Triggered RUNOUT 
insert_gcode:
  M117 Filament switch inserted
  M118 Filament Inserted 

#[filament_motion_sensor encoder_sensor]
#switch_pin: ^gpio24
#detection_length: 5
#extruder: extruder
#pause_on_runout: False
#runout_gcode:
 # PAUSE # [pause_resume] is required in printer.cfg
 # M117 Filament encoder runout
#  M118 FILAMENT encoder runout 
#insert_gcode:
#  M117 Filament encoder inserted
#  M118 FIlament encoder detected 

```

jeffctr, 2024-08-14:
`^gpio16 for switch_sensor and ^gpio29 for encoder_sensor` will work perfectly

ansi1986, 2025-03-31:
Hello,

can I also use ^gpio16 and ^gpio25 ?

Thanks