bigtreetech/smart-filament-detection-module #42: Install SFS v2 on Artillery Sidewinder X2?

Ltek, 2023-10-17:
Anyone know how to install the SFS v2 on an Artillery Sidewinder X2?

Ltek, 2023-10-27:
Not one person has installed this on an Artillery Ruby board?

GrandSam14, 2023-12-01:
Hi,

Same request, any help will be welcome to install on Artillery SideWinder X2 and Ruby board.

Very disappointing to let my fresh bought SFS V2 in a drawer till any solution to use with my X2 -_-

Ltek, 2023-12-10:
Conclusion: Doesn't work on Sidewinder X2. 

I spent many hours testing different klipper code but have not found any way to make the motion sensor will work with the X2. It always triggers immediately after the filament purge no matter what I do. I can properly enable/disable it... and the code detects the sensor but if its enabled, it always triggers after purge. If you enable it only after the layer starts it will work but it maybe false trigger, I dont know. I no longer have an X2 so cannot do anymore testing.

ovar0409, 2023-12-30:
Hi LTek, I'm just curious... how did you connect the SFS on the Ruby board? I guess Z endstop (pin ^PA0) but which other pin for the motion sensor?
Can you also describe you printer.cfg configuration?

eugeniomiro, 2025-10-02:
I have the same question. I'm using Klipper, though. I don't know your case. 

Checking the Artillery Ruby pinout [here](https://www.reddit.com/r/Artillery3D/comments/w7wb6d/artillery_ruby_12_motherboard_pinout/), I see the UART socket could be used for this, and also the EXP sockets. I haven't tried it yet, but I'll do so soon and leave a comment here when done.