bigtreetech/smart-filament-detection-module #43: BTT 2.0 with Creality 4.2.7 system board

TacII, 2023-10-23:
Hello!

I recently purchased a BTT SFS v2.0 and I have a Creality v4.2.7 system board. The manual says, “To connect to a motherboard without filament runout port, please consult our technical support.”. 

The Creality v4.2.7 system board only has a filament runout sensor (circled) which I ran successfully with the BTT Sensor v1 and Klipper.
Now, the blue light is constantly on and never blinks on the sensor when I connect one of the two cables. With the other one the sensor is just black.
 
[image]

I’ve tried using both firmware modifications for Klipper,  [filament_motion_sensor] and [filament_switch_sensor] in printer.cfg (with everything that goes beneath them). I used own a BTT v1.0 so I’m fairly familiar with the setup.
The Creality system board filament sensor PIN is called PA4 and I’ve tried “PA4” and “^PA4” with both cables. I feel like I’m missing something here since there are two cables but only one that can be connected. Please advise.a

Ltek, 2023-10-27:
I have this same problem just with the Artillery Sidewinder X2... I don't think they really support the product. They clearly do not monitor this site at all. And over a week I put in a tech support ticket I entered on their site - they have never responded.

TacII, 2023-10-28:
Believe it or not, I actually got it to work today. I needed to resolder a cable to get both sensors to work on the Ender3 v4.2.7 system board. I’ll post more details here as soon as I have time.

negativefate83, 2023-10-29:
> Believe it or not, I actually got it to work today. I needed to resolder a cable to get both sensors to work on the Ender3 v4.2.7 system board. I’ll post more details here as soon as I have time.

mate if you could, youd be a life saver
just unboxed the sfs 2.0 and now im just scratching my head questioning my life choices

TacII, 2023-10-29:
I feel you. As everyone who owns a 3D-printer, I was standing at the cliff of despair myself. So here's how I did it, for the Creality 4.2.7 system board. You need to free up one pin-pair on the system board, which you already did if you own a BL/CR-Touch, then the Z- pin-pair should be available.

The Black and Green contact (the motion sensor) should go to the Z- pin pair on the board. It's three pins wide so you need to either cut the outer part of the contact with a sharp pair of scissors (quite easy actually) or solder on a two pin contact to the black and green wires. Black is ground and should be on the left, Green is PA7 and should be on the right.

Here's what the wiring looks like from the manual.
[image]

This is where the Motions Sensor Cable goes on the system board.
[image]

The Blue and Red wire pair (Switch Sensor) goes the the filament runout port on the system board and can be inserted as normal.
[image]

Then you'll need this in printer.cfg in Klipper. Sorry I don't use Marlin anymore but I think you could probably follow the manual and use PA4 for the Switch Sensor and PA7 for the Motion sensor.

```
#Switch sensor, blue @PA4 + red VCC 3,3V
[filament_switch_sensor btt_switch_sensor]
switch_pin: ^PA4 
pause_on_runout: false
runout_gcode:
  M117 Switch Runout!
  {action_respond_info("BTT: Switch Runout detected!")}
  PAUSE
insert_gcode:
  M117 Filament inserted
  {action_respond_info("BTT: Filament inserted.")}
  
#Motion sensor, Green @PA7 +Black GND
[filament_motion_sensor btt_motion_sensor]
switch_pin: ^PA7 # Previous Z- on 4.2.7
detection_length: 2.88 # accuracy of motion sensor 2.88mm
extruder: extruder
pause_on_runout: false
runout_gcode:
  M117 Motion Runout
  {action_respond_info("BTT: Motion Sensor Runout detected!")}
  PAUSE
insert_gcode:
  M117 Filament inserted
  {action_respond_info("BTT: Filament inserted.")}

[delayed_gcode DISABLEFILAMENTSENSOR]   
initial_duration: 1
gcode:
    M117 Motion FIL. SENSOR OFF.
    SET_FILAMENT_SENSOR SENSOR=btt_motion_sensor ENABLE=0
    SET_FILAMENT_SENSOR SENSOR=btt_switch_sensor ENABLE=0
```
    
And for START you will have to add two lines more:

```
######################################################################
# Start Print and End Print
######################################################################

# Replace the slicer's custom start and end g-code scripts with
# START_PRINT and END_PRINT.

[gcode_macro START_PRINT]
variable_bed_temp: 60
variable_extruder_temp: 185
gcode:
    # Start bed heating
    M140 S{bed_temp}
    # Use absolute coordinates
    G90
    # Reset the G-Code Z offset (adjust Z offset if needed)
    SET_GCODE_OFFSET Z=0.0
   
    ## Home the printer
    G28

    # Reset Extruder
    G92 E0

    ## Move to start position
    G1 X2.1 Y20 Z0.28 F5000.0

    # Move the nozzle near the bed
    G1 Z5 F3000
    # Move the nozzle very close to the bed
    G1 Z0.15 F300
    # Wait for bed to reach temperature
    M190 S{bed_temp}
    # Set and wait for nozzle to reach temperature

    #KAMP Calibrate print area only
    BED_MESH_CALIBRATE
    
    M109 S{extruder_temp}
  
    # Move Z Axis up
    G1 Z2.0 F3000

    #KAMP Line Purge
    LINE_PURGE
        
    SKEW_PROFILE LOAD=my_skew_profile_231010
   
    _M117 BTT SENSOR ON.
    SET_FILAMENT_SENSOR SENSOR=btt_motion_sensor ENABLE=1
    SET_FILAMENT_SENSOR SENSOR=btt_switch_sensor ENABLE=1
```
For CANCEL:
```
[gcode_macro CANCEL_PRINT]
description: Cancel the actual running print
rename_existing: CANCEL_PRINT_BASE
gcode:
    TURN_OFF_HEATERS
    CANCEL_PRINT_BASE
    M117 MOTION FIL. SENSOR OFF.
    SET_FILAMENT_SENSOR SENSOR=btt_motion_sensor ENABLE=0
    SET_FILAMENT_SENSOR SENSOR=btt_switch_sensor ENABLE=0
```
And lastly for END:
```
# ---------------------------- End Print -----------------------------
[gcode_macro END_PRINT]
variable_machine_depth: 235
gcode:
    # Turn off bed, extruder, and fan
    M140 S0
    M104 S0
    M106 S0
    # Relative positionning
    G91
    # Retract and raise Z
    G1 Z0.2 E-2 F2400
    # Wipe out
    G1 X5 Y5 F3000
    # Raise Z more
    G1 Z10
    # Absolute positionning
    G90
    # Present print
    G1 X0 Y{machine_depth}
    # Disable steppers
    M84
    M117 BTT SENSOR OFF.
    SET_FILAMENT_SENSOR SENSOR=btt_motion_sensor ENABLE=0
    SET_FILAMENT_SENSOR SENSOR=btt_switch_sensor ENABLE=0
```

negativefate83, 2023-10-29:
mate, to say im grateful doesnt even come close
love your work, thankyou so much

TacII, 2023-10-29:
It seems that I haven't really got the motion sensor working fully as I thought yesterday. Sometimes it is detected and sometimes not (often it is there after a firmware restart.) But as soon as it starts printing it is empty. Not sure what is going on here. I'll dig on. At least the Switch works which is more important, so if the printer runs out of filament it'll pause the print.

negativefate83, 2023-11-04:
Hey mate how’d you go getting it sortedHaven’t even gotten around to putting my sensor in yet but the plan is to make a move todaySo what do you suggest… don’t even bother with the motion sensor and just run the switch on its own?

TacII, 2023-11-07:
~PA7 instead of ^PA7 for the motions sensor actually makes Klipper detect filament - visible under the Misc part on the dashboard but also through the command "QUERY_FILAMENT_SENSOR SENSOR=btt_motion_sensor".
But as soon as it starts printing it fails after the set "detection_length". 

From the Klipper manual :
_Input pins may be preceded by ^ to indicate that a hardware pull-up resistor 
should be enabled for the pin. If the micro-controller supports pull-down resistors then an input pin may
alternatively be preceded by ~. Pin names may be preceded by ! to indicate that a reverse polarity should be 
used (eg, trigger on low instead of high)._

I've also tried ^!PA7 and ~!PA7 which than makes the sensor detect a runout immediately but the sensor is not true. ^PA7 should be correct. The Z- pin should be a pull-up resistor after what I've read. But no setting I tried has made the blue led flash as it should when filament is passing by it. One thing that might work that I haven't tried is to switch the motion-cable and switch sensor cable but that requires new soldering of the contacts since the switch sensor cable contact has three poles and won't fit into the Z- port. 

I'm quite disappointed in BTT who haven't answered my support request two weeks ago. They're not active here or on Discord. 

[image]

Ltek, 2023-11-07:
Guys, I've been trying to get this working on an Artillery Sidewinder X2 (running Klipper) -- I suspect we have similar or identical problems. It does not detect motion for me, and Pauses immediately after purging the nozzle.

The X2 only has 1 Z-Stop connector I can use for the Motion Sensor, thus the SFSv2's Runout wire/pin is not connected.

Is there code we can use to test the filament sensor without printing... just moving the filament in/out by hand?

shaidou1287, 2023-11-10:
So, I got this working on a Creality 4.2.7 board. What I did was just set the PIN to the Z-Stop pin PA7, no pull-up or pull-down. I have tested it multiple times mid-print and it will pause the print and wait for me to resume it. Hope this helps those who are having issues. BTW this was on Klipper.

TacII, 2023-11-10:
But have you got the motion sensor to work? The blue light should be blinking while you print.

shaidou1287, 2023-11-10:
Yeah, the blue light is blinking and if I hold the extruder open so filament won't go through it runs my pause macro.

TacII, 2023-11-11:
Wow, interesting! Can you please share the details? So motion sensor cable (green+black) to PA7, right? And blue+red switch sensor cable to PA4 I assume? Or aren't you using the switch cable at all? 

Can you please drop in your printer.cfg here so that I can compare the differences? Because I tried all combinations of Pull up/down.

Thanks!

negativefate83, 2023-11-11:
Interesting…

shaidou1287, 2023-11-14:
Sorry for the Delay but here is the section of the Printer config I have for the sensor. The rest is just the basic Ender 3 V2 config from the example w/ a CR Touch added.

`# Runout Sensor #########################################################################
[filament_switch_sensor switch_sensor]
switch_pin: PA4
pause_on_runout: False
runout_gcode:
  PAUSE # [pause_resume] is required in printer.cfg
  M117 Filament switch runout
insert_gcode:
  M117 Filament switch inserted

[filament_motion_sensor encoder_sensor]
switch_pin: PA7
detection_length: 2.88
extruder: extruder
pause_on_runout: False
runout_gcode:
  PAUSE # [pause_resume] is required in printer.cfg
  M117 Filament encoder runout
insert_gcode:
  M117 Filament encoder inserted`

Egosumumbravir, 2023-11-15:
> This is where the Motions Sensor Cable goes on the system board.

Crap, what if you're already using the PA7 pin to get a Biqu MicroProbe v2 to work?

Does it need hardware pullup? The MicroProbe V2 does which is why it has to goto PA7 instead of the default touch port PB1.

I wonder if PB1 could be used in this case?

shaidou1287, 2023-11-15:
Honestly I have absolutely no clue. All I know is that it works with out the pull up or pull down and the way I got to thinking about this is that the v1.0 doesn't use pull up or pull down for its encoder.

Egosumumbravir, 2023-11-15:
> All I know is that it works with out the pull up or pull down...

That'll be the key point - the Microprobe needs a hefty pullup to function correctly which PB1 lacks. Wiring is gonna get a bit messy but it'll be cool if it enables full functionality.
I'm compiling MRiscoC Professional Firmware for it. Will report back once I've had a chance to fiddle with it.

sdmorrey, 2023-12-20:
I’m about to try this out thanks so much for your hardwork everyone!

Just a quick tip, I know the default is to want to break out the soldering iron and solder these things but it makes a mess.  

With these “pinned connectors” all you have to do is repin them.  It’s super easy, just grab a pair of tweezers, poke the pins latch on the flat side until it’s down a bit and remove the wire.  Then use your fingernail or something to pry the latch flap back about 45 degrees and stuff it into the hole on the other connector.  Repeat until there’s no more pins.

sdmorrey, 2023-12-20:
Since PA7 is the old z- end stop which you no longer need if you have a bltouch, wouldn't it make more sense to make that one your runout sensor?

Then PA4 could be used for the motion encoder since that's not just a simple switch but a lightbulb and sensor counting optical interrupts.

I had them switched up (because I was doing it by myself before finding this thread, and had to swap them in the code.

Egosumumbravir, 2023-12-20:
> Since PA7 is the old z- end stop which you no longer need if you have a bltouch, wouldn't it make more sense to make that one your runout sensor?

It would *if* I had a BL or CR-Touch unit. I dont. I'm rocking a v2 BIQU Microprobe which doesn't work in the standard PB1 probe port because the pullup there is insufficient. So the MPv2 goes into PA7 and PB1 is free. Hence breaking out the soldering iron to make a splitter to connect the two ports.

Seems to be working so far as a simple presence detector. Haven't yet got around to compiling a custom Marlin but I think that will be an either/or situation - Marlin only supports one runout sensor per extruder AFIK.

danderik88, 2023-12-26:
@TacII im so gratefull someone tryed this out before. I wanted to try it today and i made my last search on google to find out if this was already done before i make some mistakes. Thank you so much!

ShaunTheBaa, 2024-02-18:
> Marlin only supports one runout sensor per extruder AFIK.

I didn't realize this was the case. I only just bought the BTT V2.0 run-out sensor for my Creality 4.2.7 board. I'm right now trying to figure out how to make it work in Marlin but looks like it might not be possible.
I do see the option #define FILAMENT_MOTION_SENSOR but there is no pin assignment. It looks like there was in a previous Marlin version. Nevertheless, if someone figure out how to make it work in Marlin, let me know what you did please.

BakoBangBang, 2024-04-02:
Hello, I am trying to add the BTT SFS V2.0 with the command lines mentioned above but I am not having any luck. Could anyone get it to work? Ender 3 V2

shaidou1287, 2024-04-02:
> Hello, I am trying to add the BTT SFS V2.0 with the command lines mentioned above but I am not having any luck. Could anyone get it to work? Ender 3 V2 whit klipper

What command lines are you using?

BakoBangBang, 2024-04-02:
> > Hello, I am trying to add the BTT SFS V2.0 with the command lines mentioned above but I am not having any luck. Could anyone get it to work? Ender 3 V2 whit klipper
> 
> What command lines are you using?

The ones you uploaded... I already found the error and it worked (I was placing the lines in the start and end gcode)... Is there a way to configure the times in which the motion sensor is activated? Since sometimes the head does travel too far to the edge of the bed when returning the filament that was stretched too much, it makes the sensor think that it is stuck

shaidou1287, 2024-04-02:
That is totally separate thing in klipper. it's actually a klipper command
you can run to enable and disable the sensor. That would be something you
would need to put into code.

SET_FILAMENT_SENSOR¶
SET_FILAMENT_SENSOR SENSOR=<sensor_name> ENABLE=[0|1]: Sets the filament
sensor on/off. If ENABLE is set to 0, the filament sensor will be disabled,
if set to 1 it is enabled.

Is the command reference from klipper documentation.

clove0, 2024-08-22:
Thought I'd chime in here as I did get this working with Marlin.

You have to be running MARLIN-BUGFIX-2.1.X for this to work. If you're apprehensive about migrating, there's a feature within VSCODE that lets you compare files so you can move your changes over with relative ease. (Right click first file "select for compare", right click second file "compare with selected"). 

I followed the wiring diagram shown in the manual (see original post by TacII) with a few differences:
On the 4.2.7 board, I plugged Blue/Black/Red and plugged Green into the Z- jack as illustrated in TacII's post. 

In Marlin I followed the V2 instructions shown [here](https://github.com/bigtreetech/smart-filament-detection-module) with the following changes:

 #define FILAMENT_RUNOUT_DISTANCE_MM 20
and 
#define FIL_MOTION1_PIN    PA7

I find that the filament runout distance of 5mm is too small...if the printer head moves far enough, a bunch of slack is pulled on the filament. The result is no filament length is pulled through the SFS resulting in a false runout error. 

As for the normal runout config:

  #define FIL_RUNOUT_PIN PA4
  #define FIL_RUNOUT_STATE     LOW        // Pin state indicating that filament is NOT present.
  #define FIL_RUNOUT_PULLUP 

Hope this helps someone out there!

paganinihot, 2024-09-16:
How did you guys get this thing to eventually work on 4.2.7? What was the final solution? I connected red-blue to PA4 (**blue** on the left), black-green to PA7 (**black** on the left), in Klipper I prescribed **pa4** as switch, **pa7** as motion detector. I tried it both with and without the ^ sign. During normal printing the blue indicator blinks. If the filament runs out, the PA4 detector is triggered and the printer goes into pause mode. If you open the extruder and prevent the filament from moving, the blue light stops blinking and turns off. But the printer does not pause! 

One more observation: after rebooting Klipper, if no filament is inserted, both sensors are in not detected/empty status. As soon as I insert a filament, both sensors go into "detected". And then only the filament sensor is triggered, the status of the motion sensor never changes. 

Guys, what should I write in the settings? Actually, have 4.2.2, but (I mean) its similar as 4.2.7

clove0, 2024-09-16:
I got it working with Marlin no problem (see comment above). One thing to note that came to light later one, the circular slider switch on the front of the sensor sometimes needs to be tightened. If it comes loose, you'll get a false (binary) filament runout detection. Could see this driving someone nuts who's not sure if they even have the pin assignment done correctly in klipper/marlin

Make sure the slider switch button head is finger tight when rotated clockwise.

paganinihot, 2024-09-16:
> I got it working with Marlin no problem (see comment above). One thing to note that came to light later one, the circular slider switch on the front of the sensor sometimes needs to be tightened. If [...]

But my problem is motion sensor not working (for klipper, I mean, coz blue lamp blinking during normal printing, so, signal from BTT sending to board and... klipper not received it?) and not enetering in pause mode when filament not moving (but present). Works only switch present/not present filament.

Egosumumbravir, 2024-09-16:
> How did you guys get this thing to eventually work on 4.2.7?

I never got it working properly on Marlin (single filament sensor limitation) and for other reasons needed to change out the 422 board for a SKR Mini E3 V3 alongside a switch to Klipper.  Works great there - both presence and motion will pause the machine and wait for interaction.

The motion sensor only applies when the extruder moves and there's not linked movement from the sensor. It's a little unintuitive but does work. I'd be double checking klipper configs if these are not functioning as expected.

> One thing to note that came to light later one, the circular slider switch on the front of the sensor sometimes needs to be tightened.

Same. A drop of blue Loctite would have been a great factory addition.

merlinsvq, 2024-10-28:
Graciias Tacll, funcioona en Marlin y Klipper en una ender 3 con placav4.2.7, mi errror era que se habia soltado un pin del conector que va a la placa base. De nuevo, muy agradecido

BeardedViking1488, 2024-11-11:
ok so im really struggling here. Pretty new to using VSCode and have done some basic work in arduino IDE. But i cannot for the life of me figure out how to compile this firmware. 

Ive edited my configuration.h file, ive followed all the steps and have the extensions loaded etc etc but its just not working for me - dont suppose anyone can upload  a .bin file of their firmware for me? im pulling my hair out here!!! lol

pbussman, 2024-11-30:
Just wondering does anyone have a config file for an Ender 3 pro with a CR touch and the BTT filament runout sensor that they would be willing to share?

ArjanMal, 2024-12-05:
> Just wondering does anyone have a config file for an Ender 3 pro with a CR touch and the BTT filament runout sensor that they would be willing to share?

If so i would be very interested also!

Falconus, 2024-12-09:
I got it working with the stock Creality **4.2.2** board in Marlin. If you haven't read the previous posts, note that you will have to have a z-probe, because you need to free up the PA7 port (by default, for the Z-stop switch) if you want to have **both** the runout sensor and motion sensor working. I am not going to share my *entire* config files right now (I am running a BL-touch, a Microswiss NG direct drive extruder with a BMS Dragonfly heater block, a 60W heater and a P1000 thermistor, so I don't think it would be too helpful for anybody else to build my config).

Anyway, it definitely seems to be sensing when I don't load the filament, and it successfully paused the print yesterday when my heat break got jammed with nylon (that wasn't fun to fix by the way).

Apologies for the weird formatting, the code tags aren't working for me today.

Thank you **everyone** in this thread; I wouldn't have figured this out without the various steps and ideas that you outlined above.

Here is what I ended up doing:

### Physical connections ###
1. repin the wires on the motherboard side of the cable so that the *blue* wire and the black ground wire go to *PA7* (you'll need a J2 connector). Make sure you don't switch the ground and the blue wire (see @TacII 's diagram above). Insert the green wire where the blue wire had been originally.
3. Double-check that the red and black wires are where they are supposed to be, to avoid damaging stuff.
4. **Note that I could only get it to work with the blue wire going to PA7, and the green wire going to PA4.**

### Marlin ###
#### File 1: pins_CREALITY_V4.h ####

1. Comment out Z_STOP_PIN definition
//
// Limit Switches
//
#ifndef X_STOP_PIN
   #define X_STOP_PIN                        PA5
   #endif
#ifndef Y_STOP_PIN
  #define Y_STOP_PIN                        PA6
#endif
//#ifndef Z_STOP_PIN
  //#define Z_STOP_PIN                        PA7
//#endif

2. Uncomment FIL_RUNOUT_PIN and set to PA7
//
// Filament Runout Sensor
//
#ifndef FIL_RUNOUT_PIN
  #define FIL_RUNOUT_PIN                    PA7   // "Pulled-high"
#endif

#### File 2: Configuration.h ####

##### Filament runout sensor component #####
1. Uncommment #define FILAMENT_RUNOUT_SENSOR
2. set #define FIL_RUNOUT_PIN PA7
3. #define FIL_RUNOUT_STATE HIGH
4. **comment out #define FIL_RUNOUT_PULLUP** - note that this is contrary to the [instructions that BigTreeTech provides] (https://github.com/bigtreetech/smart-filament-detection-module)

##### Filament Motion Sensor Component #####
1. set #define FILAMENT_RUNOUT_DISTANCE_MM to whatever value makes sense for your application
2. uncomment #define FILAMENT_MOTION_SENSOR
3. uncomment #define FILAMENT_SWITCH_AND_MOTION
4. set #define FIL_MOTION1_PIN PA4
5. **set #define FIL_MOTION1_STATE LOW** -again, this is opposite of the official instructions
6. set #define FIL_MOTION1_PULLUP

##### Other #####
Per the instructions: 

> The filament sensor feature of Marlin also needs to enable #define NOZZLE_PARK_FEATURE and #define ADVANCED_PAUSE_FEATURE. please uncomment #define NOZZLE_PARK_FEATURE and #define ADVANCED_PAUSE_FEATURE and set specific parameters according to your printer.

enable the nozzle park feature.

/**
 * @section filament runout sensors
 *
 * Filament Runout Sensors
 * Mechanical or opto endstops are used to check for the presence of filament.
 *
 * IMPORTANT: Runout will only trigger if Marlin is aware that a print job is running.
[... 101 lines cut ...]

        //#define FIL_MOTION6_STATE LOW
        //#define FIL_MOTION6_PULLUP
        //#define FIL_MOTION6_PULLDOWN

        //#define FIL_MOTION7_STATE LOW
        //#define FIL_MOTION7_PULLUP
        //#define FIL_MOTION7_PULLDOWN

        //#define FIL_MOTION8_STATE LOW
        //#define FIL_MOTION8_PULLUP
        //#define FIL_MOTION8_PULLDOWN
      #endif
    #endif // FILAMENT_MOTION_SENSOR
  #endif // FILAMENT_RUNOUT_DISTANCE_MM
#endif // FILAMENT_RUNOUT_SENSOR

#### File 3: Configuration_adv.h ####
1. Enable #define HOST_ACTION_COMMANDS
2. Enable #HOST_PROMPT_SUPPORT
3. Per the instructions:

>The filament sensor feature of Marlin also needs to enable #define NOZZLE_PARK_FEATURE and #define ADVANCED_PAUSE_FEATURE. please uncomment #define NOZZLE_PARK_FEATURE and #define ADVANCED_PAUSE_FEATURE and set specific parameters according to your printer.

enable the Advanced Pause Feature.