bigtreetech/smart-filament-detection-module #46: SFS 2.0: filament path, a design flaw - filament hits walls near output holes

arekm, 2023-12-04:
That's one big problem with SFS2.0 filament sensor. Pain to use.

Current design of output holes in filament path is like this:

[image]

Just a small diameter hole. That filament needs to hit in perfectly.

In my experience, especially when having longer PTFE tubes installed on both ends of SFS2.0 sensor it is almost impossible to push filament out via that hole. That's because even tiny curvature of filament (by that I mean just filament taken from spool - it won't be straight, it will take a shape of spool) or side tension on filament causes it to hit a wall on left/right side of that exit hole. 

I have to unplug PTFE tube on "entry" side and manoeuvre filament blindly inside SFS2.0 sensor, so it finally hits the exit hole of the sensor. It's pure pain to do that each time.

Perhaps a funnel shape would be better, directing the filament from the edges and directing it towards the center of the exit hole.

[image]

Consider such improvement on 2.1 iteration.

(In theory we could just improve and print own, better case for it but CAD model is broken (https://github.com/bigtreetech/smart-filament-detection-module/issues/45))

osvch, 2024-04-19:
I'm facing this issue heavily- I'm happy that it was already posted.
@bigtreetech what would be mitigation plan for the users?

osvch, 2024-05-18:
> +1

> +1

I tighten the PTFE connectors, and before filament insertion- I straighten it as much, as I can- that helps a let with filament transition.

osvch, 2024-05-18:
@2Balu2 @roykrikke 
I tighten the PTFE connectors, and before filament insertion- I straighten it as much, as I can- that helps a let with filament transition.

noobydp, 2024-07-05:
Try rotating the sensor 180 degrees, and shove the filament in the other direction.

LsHallo, 2025-01-13:
Do you thinks its possible to disassemble the sensor and use a small drill bit to kind of create this funnel shape?
This sensor is driving me mad every time i have to insert filament...

osvch, 2025-01-13:
I got update on my sensor and  I was able to address all issues:

- Circle part "button" on the sensor front side fell off (marked red at the photo), I disassebmled sensor, tighten everything again
- I rotated my sensor to infeed filamnet horizontally (cable connector facing down)
- Make sure at least 4-5cm of filament you are inserting is straight
- Tighten the PTFE tube connectors

After those steps I do not see issues with inserting the filament like before.

[image]

2Balu2, 2025-01-13:
Thanks. I did use a dremel to create a smoother transition/bigger inlet.

ljohnso16, 2025-11-11:
> Thanks. I did use a dremel to create a smoother transition/bigger inlet.

any more details, so i can do that too ?