bigtreetech/smart-filament-detection-module #47: Creality K1? Can the sensor be used if only filament runout pins are available?

Horgul123, 2023-12-29:
The Creality K1 seems to have no spare port for the motion detection.
Is there a way to use the sensor with this printer?
Or in general:
Is there a way to use the sensor if only a "normal" filament runout sensor can be installed? (e.g. sensor triggers the filament runout if there is a filament runout OR motion issue)

The-last-phantom, 2024-01-18:
Big Tree Tech will tell you the K1 cant support the SFS 2.0, but I can confirm the SFS 2.0 works perfectly on the K1 Max. On the standard K1, you might have to just lose the switch and only use the motion sensor. On the Max you can use both.

It's simply a case of de-pinning the SFS connectors, and repinning them to suit the K1. The connector bodies are too big for the ports on the motherboard, you would either need to get smaller connectors or I simply put a piece of heat shrink over each pin so they dont accidentally touch each other. The Max has the second filament sensor port on the toolhead PCB, so this can be used to get both sensor signals.

On the Max motherboard (I imagine this is the same for the standard K1, but I would still double check), the 5V VCC comes from the pin closest to the WIFI antenna, the GND is the centre pin, and the signal is the last pin. So simply de-pin the SFS connectors, and plug the red wire onto the pin closest to the WIFI antenna, the black wire onto the centre pin, and the green wire onto the last pin. If you want to use the blue wire to get the runout switch, you would need to separate it from the other wires and run it to the toolhead PCB - I simply folded it back and left it unconnected. Then the SFS 2.0 harness is the perfect length to run back the exact same route as the stock filament sensor harness, and the sensor body *just* fits where the stock sensor sits with a custom bracket (the extruder can make contact with the bulge on the body here, and you want to be careful to not sit the sensor on the belt). If you want to mount it externally, you will probably need to extend the harness a bit.

The rest of the install is as per the manual, make the changes to your printer.cfg file. If you've wired it as I've described, then your motion sensor is on pin PC15. 

All up, took me about two hours to install including the time to work out the pinouts on the motherboard. Super easy.

Horgul123, 2024-01-19:
Thanks a lot for this detailed explanation! 
So just for my understanding: 
It doesn't detect a filament runout via the switch but the printer will check if there is a filament movement when there should be one. And if there is no movement (no filament or filament blocked) it will raise a "out of filament" error.

The-last-phantom, 2024-01-20:
Yeh, correct. The motion sensor alone can do the same task as the switch - if the filament runs out or breaks, the motion sensor will trigger. If the filament jams in the extruder or snags at the spool, the motion sensor will trigger. 

The only somewhat downside is that it may sometimes trigger a false positive, which tends to happen in parts of the model where the printer is doing a lot of retractions and the sensor doesn't pick them up. This can be tuned out by playing with the detection length, but it means spending a bit of time testing. At the moment, I haven't had a false positive since raising the detection length from the minimum 2.88mm to 3.5mm, but I've only done about 2.5 hours of printing since I made that change. 

I am also using this in conjunction with Henlor's extruder runout sensor mod that you can find on Printables, which puts a micro switch into the extruder housing that then plugs into the port on the toolhead PCB. The reason I like this instead of just using the BTT micro switch, is it means the printer will only detect a runout right at the toolhead rather than at the back of the printer enclosure. It also means I can enable automatic filament loading, so the printer will automatically load and extrude a defined length of filament when the switch is triggered. And finally, it means I can bypass the reverse bowden tube when using TPU, which would mean that I have to disable the motion sensor as there is no filament running through it. This then provides a bit of safety for having a runout while using TPU.

Selberg666, 2024-01-31:
Hy guys!
I just bought a SFS 2.0 and want to install it on K1 Max.
Sooo if i connect the blue wire to the signal pin from the tool head it will activate the switch from SFS?
I don t mind to run 1 wire to the tool head....
Also.....found out on K1 the middle pin is signal and the last pin is ground......so i heard from someone that put SFS 1 on a K1.

Selberg666, 2024-01-31:
If the tool head pin is PA10...
Then the klipper line to add is this?

[filament_switch_sensor switch_sensor]
#BLUE WIRE ON TOOL HEAD
switch_pin: PA10
pause_on_runout: True
runout_gcode:
  PAUSE # [pause_resume] is required in printer.cfg
  M117 Filament switch runout
insert_gcode:
  M117 Filament switch inserted

[filament_motion_sensor encoder_sensor]
switch_pin: PC15
detection_length: 2.88
extruder: extruder
pause_on_runout: True
runout_gcode:
  PAUSE # [pause_resume] is required in printer.cfg
  M117 Filament encoder runout
insert_gcode:
  M117 Filament encoder inserted

Horgul123, 2024-02-03:
@The-last-phantom I have some problems with the config.
Could you please show me your entry in the printer.cfg? Did you do any additional changes?

I tried to copy the things from the BBT manual but I get:
`{"code": "key342", "msg": "Option 'detection_length' is not valid in section 'filament_switch_sensor filament_sensor'\n\nOnce the underlying issue is corrected, use the \"RESTART\"\ncommand to reload the config and restart the host software.\nPrinter is halted\n", "values": ["detection_length", "filament_switch_sensor filament_sensor"]}`

Selberg666, 2024-02-03:
I had succes with this line 

[filament_motion_sensor encoder_sensor]
switch_pin: !PC15
detection_length: 3.2
extruder: extruder
pause_on_runout: true
runout_gcode:
  {% if printer.extruder.can_extrude|lower == 'true' %}
    G91
    G0 E30 F600
    G90
  {% endif %}

[filament_switch_sensor filament_sensor_2]
pause_on_runout: true
switch_pin: nozzle_mcu:PA10
runout_gcode:
  {% if printer.extruder.can_extrude|lower == 'true' %}
    G91
    G0 E30 F600
    G90
  {% endif %}

the mount for the sensor i designed it myself...here is a link
https://www.printables.com/model/747290-bigtreetech-sfs-v20-bracket-mount-on-creality-k1-m

Horgul123, 2024-02-03:
Thanks a lot!
Will try your code :) And thanks for the mount. Will print it tomorrow and check if it works on a K1 (not max)

Selberg666, 2024-02-03:
it should work with k1 but you will have to drill a hole in the k1's back to put the screw in.
In the k1 max you just remove the back filament guide and you have he hole.

Horgul123, 2024-02-03:
ok. Thanks. Will check if I find a way without drilling a hole :)

Horgul123, 2024-02-04:
@Selberg666 : I've added a comment at printables. :)  I was able to install it, but need to do some modifications to avoid drilling a hole in the K1.

And a question:
Is there a trick to push the filament through the sensor? I have "some" problems pushing it through....

Selberg666, 2024-02-04:
Yeahhhhh straighten the filament ad much as possible and then try to insert it......major design flaw in the sensor.....if the filament isn t straight it goes sideways after you get past the switch wheel in the sensor.

Selberg666, 2024-02-04:
How it is with the code line..???...i mean it works fine but i still have the filament missing notification on the printer display and can t make it go away.
The printer pauses the job at filament end and jams but i don t like that error on the display.

Horgul123, 2024-02-04:
> Yeahhhhh straighten the filament ad much as possible and then try to insert it......major design flaw in the sensor.....if the filament isn t straight it goes sideways after you get past the switch wheel in the sensor.

Ah ok.. This could be issue... There was only little filament left so it was not very straight.
A new spool worked fine.

Need to check what happens when I have the next filament error..

Selberg666, 2024-02-04:
solved the problem....new code line....this one works perfect

[filament_switch_sensor filament_sensor]
pause_on_runout: true
switch_pin: nozzle_mcu:PA10
runout_gcode:
  {% if printer.extruder.can_extrude|lower == 'true' %}
    G91
    G0 E30 F600
    G90
  {% endif %}

[filament_motion_sensor encoder_sensor]
switch_pin: PC15
detection_length: 3.2
extruder: extruder
pause_on_runout: true
runout_gcode:
  {% if printer.extruder.can_extrude|lower == 'true' %}
    G91
    G0 E30 F600
    G90
  {% endif %}

Horgul123, 2024-02-04:
in my cgf I have
switch_pin: !PC15

You don't invert the signal. I would expect that if you use the wrong version, it immediatly stops and throws an error.... strange

Selberg666, 2024-02-04:
With ! I would get instant pause print and filament runout after calibration end... that is why i inverted the signal.

Horgul123, 2024-02-04:
Ah ok. I'll test it as I'm getting many pause during the current print

Horgul123, 2024-02-05:
Changed my settings but still getting false positives :(

Selberg666, 2024-02-05:
That is why you need both sensors connected.....they work together somehow.
Mine is working flawless

The-last-phantom, 2024-02-05:
You don't need both sensors at all, they don't interact with each other in any way.

Play around with your detection length setting. 3.2mm is very low and can trigger a lot of false positives depending on your filament and print settings. For example, printing something with a lot of retractions in a small area will mean that your printer can print quite a lot of filament, but the sensor never actually detects more than 3mm because it's doing lots of 1.5mm movements. That will then flag an error when there is none. 

Increase your detection length to get more reliable results.

Horgul123, 2024-02-05:
@The-last-phantom at the moment have 4.5mm and still a stop in the middle of the night.
My fear is: if I increase it too much a real error could be noticed too late :(

The-last-phantom, 2024-02-05:
4.5mm is still very low. That said, it's actually the same as what I am using, and I only very occasionally get a false positive. It becomes very dependant on what you are printing, and you might need to adjust your detection length to suit the model you're printing. You just can't print a model with loads of retractions and have a very low detection length. And if you really want to, you might have to accept the fact that you'll get the odd false positive as a trade-off for the higher security against actual printing errors. Just keep in mind that before the SFS 2.0, most optical motion sensors were using a detection length of around 7.0mm for reliability. Generally that is still going to be low enough to save your print in the event of an actual error.

One other thing to check is just how securely your sensor is mounted. If it has any play in it, then that can actually translate to the sensor not detecting the motion because the whole sensor moves instead of just the filament. Unlikely if you've mounted it securely, but something for anyone reading to keep in mind.

Selberg666, 2024-02-05:
Sooo.....after i conected the switch sensor to the tool head and rewrite the code i didn t have 1 false reading.
The sensor was mounted in the exact same position
I printed PLA,PETG,ASA.....i even printed a small keychain with TPU and i didn t have a problem 
Diffrent prints with diffrent retractions.
And yes...the sensor works...i had 2 spool tangles and 1 filament runount.
Befor the switch sensor conect i had a lot of false detections.
Soooo just saying........do what you want...i just share my experience
IF the 2 sensors are not interacting with each other then why did they put them?Why not put just the encoder......that can do both tasks?I think there is a reason why!!!!

The-last-phantom, 2024-02-05:
Yes, there is a reason, but it's much less complex than what you are thinking. The reason is simply that not everyone wants to pause in the event of a motion error, some people want just a notification. Which is simply down to the fact that motion sensors will invariably give more false positives, it's just the nature of how they work. So some people want  a simple runout sensor that won't throw false positives to trigger a pause in the event of an actual runout, and a simple notification in the event of a motion issue. 

The SFS V1.0 did have motion detection only, which also worked fine. They've obviously taken feedback onboard for the V2.0 and included a more simple switch.

Horgul123, 2024-02-05:
Ok. Thanks for the explanation. Then I will try to increase the detection length a little bit more.

nickboul, 2024-03-06:
Can I connect the sfs where the stock filament sensor is connected?

Horgul123, 2024-03-06:
> Can I connect the sfs where the stock filament sensor is connected?

Yes, but you can one use the runout OR the motion sensor, as you have only one port. I use the motion sensor and it works pretty well. Sometimes a false alarm, but not very often

The-last-phantom, 2024-03-06:
Yes, exactly. Pinouts are mentioned above.

anaegeli, 2024-04-14:
Hi all,
i was wondering, what connector did you use to connector on the mainboard and the toolhead. 
Can you please tell me, so i can order the right connectors. 
Thanks

The-last-phantom, 2024-04-14:
The toolhead uses 1.25mm pitch JST connectors.

The mainboard uses different connectors. I ended up just depinning the BTT
sensor connectors and plugging the pins directly onto the board without the
connector.

ViniSug, 2024-04-28:
> Hi all, i was wondering, what connector did you use to connector on the mainboard and the toolhead. Can you please tell me, so i can order the right connectors. Thanks

@anaegeli 
K1 motherboard connector: **JST PH 2.0 3 pins**
K1 toolhead connector: **JST 1.25 3 pins**
BTT SFS v2 connector: **JST PH 2.0 4 pins**

BTT SFS cable: One side JST PH 4 pins, other side: dual JST XH 2.54 3 pins

anaegeli, 2024-04-28:
@ViniSug  
thanks a lot!

nickboul, 2024-09-12:
> > Can I connect the sfs where the stock filament sensor is connected?
> 
> Yes, but you can one use the runout OR the motion sensor, as you have only one port. I use the motion sensor and it works pretty well. Sometimes a false alarm, but not very often

And what lines should I use for my case?

nickboul, 2024-09-30:
I have tried every guide and cfg line I have found but I have not managed to make it work. Can anyone help me?

The-last-phantom, 2024-09-30:
Yep, how have you configured it and what is it doing?

nickboul, 2024-10-01:
I connected the green wire from the sfs to the existing data cable of the stock filament sensor.

Moreover I changed the following on the cfg:

[filament_motion_sensor SFS_motion_det]
switch_pin: !PC15 
detection_length: 4
extruder: extruder
pause_on_runout: true

ps: I also tried without the ! on the pc15, with the runout gcode and 7 and 14 on the detection length.

On the wiring side I have the green on the data pin, ground on the middle and power on the last pin.

nickboul, 2024-10-04:
I found it. I had to reverse the pc10 pin. My working lines are.

[filament_switch_sensor filament_sensor]
pause_on_runout: True
switch_pin: PC15
runout_gcode:
  {% if printer.extruder.can_extrude|lower == 'true' %}
    G91
    G0 E30 F600
    G90
  {% endif %}

[filament_motion_sensor encoder_sensor]
switch_pin: nozzle_mcu:PA10
detection_length: 14
extruder: extruder
pause_on_runout: True
runout_gcode:
  {% if printer.extruder.can_extrude|lower == 'true' %}
    G91
    G0 E30 F600
    G90
  {% endif %}

You can correct your guide.

Thank you

FaintandFuzzies, 2025-05-25:
Can anyone help by providing the printer.cfg coding for a Creality K1 on the SFS V2.0.  One the K1, its not connected at the toolhead... its direct to the mainboard.  Thanks.