bigtreetech/smart-filament-detection-module #48: How to install sfs 2.0 on a spider 2.3

llambion72, 2024-01-11:
I have a siboor voron 2.4 kit that it comes with spider 2.3 mcu. I bought a btt sfs 2.0.
Have anyone installed sfs 2.0 on this board?
Can anyone help me with mcu wiring?

m3cinc, 2024-03-30:
Have a look at https://www.youtube.com/watch?v=FNqQoJazBSE. In terms of adapting to the spider board, I have used on a v3.0 (similar to the v2.3 version) the X-MAX and Y-MAX inputs, since I use on my Voron 2.4R2 350 CAN and sensorless homing for the X and Y axes.
For Klipper configuration, I included sfs2.cfg in printer.cfg and placed in sfs2.cfg the following code:

[filament_switch_sensor BTT_sfs2]
switch_pin: ^PA1								
# X-MAX on Spider v3.0
pause_on_runout: False
# When set to True, a PAUSE will execute immediately after a runout
# is detected. Note that if pause_on_runout is False and the
# runout_gcode is omitted then runout detection is disabled. Default
# is True.
runout_gcode: 
#  PAUSE # [pause_resume] is required in printer.cfg
   M600
insert_gcode: 
   LOAD_FILAMENT
event_delay: 3.0
# The minimum amount of time in seconds to delay between events.
# Events triggered during this time period will be silently
# ignored. The default is 3 seconds.
pause_delay: 0.5
# The amount of time to delay, in seconds, between the pause command
# dispatch and execution of the runout_gcode. It may be useful to
# increase this delay if OctoPrint exhibits strange pause behavior.
# Default is 0.5 seconds.

[filament_motion_sensor Encoder_sensor]
switch_pin: ^PA2
# Y-MAX on Spider v3.0
detection_length: 5
# anything over 2.88 per specs
extruder: extruder
pause_on_runout: False
runout_gcode:
   PAUSE
   M117 Filament Move Error Detected
insert_gcode:
   M117 Filament Move Restored

I believe you also need somewhere in printer.cfg a [pause_resume] block.

[pause_resume]
recover_velocity: 50.
#   When capture/restore is enabled, the speed at which to return to
#   the captured position (in mm/s). Default is 50.0 mm/s.

Hope this helps!

llambion72, 2024-04-10:
Thank you !! I'll try it

eugenim-atmega, 2024-04-12:
Hello! 
Did you manage to install this device? 
Were there any problems with connection and setup? 
And also, X-MAX and Y-MAX, what is the voltage? Because for this device it is written that no more than 5 volts.
Thank You for answers.

eugenim-atmega, 2024-04-12:
OMG!
When I opened my eyes ))) look at this schem, i understand information about voltage. Its 3.3 volts. Sorry for stuped questions!
https://wiki.fysetc.com/images/Spider_V2.3_Pinout.jpg