bigtreetech/smart-filament-detection-module #60: BTT SFS 2.0 on rooted K1 Max in place of original sensor? Keeps throwing an error.

cmcfalls2, 2024-08-11:
Can someone help me with setting up a BTT SFS 2.0? 

The following code "works" in the sense that I can print, but I keep getting false runouts causing the print to stop. I had to use the PC15 (rather than !PC15) in order to get the sensor to detect filament. It literally says filament is detected, but it still pauses the print for a runout. If I use !PC15 then it shows no filament detected, even though there is filament in the printer. Here's the working code:

[filament_switch_sensor filament_sensor]
switch_pin: PC15
pause_on_runout: true
runout_gcode:
  {% if printer.extruder.can_extrude|lower == 'true' %}
    G91
    G0 E30 F600
    G90
  {% endif %}

However, if I change the code to the following, then I get an error that states "pin PC15 used multiple times in config"

[filament_motion_sensor encoder_sensor]
switch_pin: PC15
detection_length: 3.2
extruder: extruder
pause_on_runout: true
runout_gcode:
  {% if printer.extruder.can_extrude|lower == 'true' %}
    G91
    G0 E30 F600
    G90
  {% endif %}

Why am I getting that error? I do NOT have multiple instances of PC15 in my config file. I've checked using a search. And besides, if I did, I would get that same error with the original code, but I do not. I'm just trying to use the sensor as a motion sensor (as opposed to a detection sensor) since no motion would equate to a runout or a clog.

Why the crap can't I get this sensor to work properly? I worked fine before I had to do a reset, and now I can't get it working!

lukasmejia, 2024-09-25:
if you have a rooted machine access ssh and run the helperscript command and make sure you don;t have installed #14 (M600 Support), remove it if so and check again with !PC15. also comment out this part on config.cfg: 

[filament_switch_sensor filament_sensor]
switch_pin: PC15
pause_on_runout: true
runout_gcode:
{% if printer.extruder.can_extrude|lower == 'true' %}
G91
G0 E30 F600
G90
{% endif %}

ducati1198, 2025-02-05:
your using same pins for 2 different sensor better seperate them