bigtreetech/smart-filament-detection-module #62: SFS v2.0 + SKR 1.4 Turbo + TFT35 E3 V3.0 How do I get work together if I print only from an SD-card on TFT35?

nen777w, 2024-08-21:
Not recently I purchased SFS V2.0 to avoid situations when filament can get stuck, or nozzle clogging.

Before that, I had experience using a simple filament sensor. I plugged it into TFT35 directly (not in SKR board) and this was the only way to get it working.
Because I print from TFT35, SD-card slot only.

TFT35 has only 3 pin slots for the Filament sensor,  but in the case of SFS v2.0 it has 4 pins.
The BTT recommends plugging it directly into the SKR, but I'm afraid that it will not work because as I mentioned above I print from the TFT35 SD-card slot. M'i right?

My question is: 
Would that work for my usual printing case?

kimeschris, 2025-01-06:
Have you been able to get it working? I have the exact same setup and can't figure out how to make it work.

nen777w, 2025-06-16:
> Hello, I have this same detector but on the 70 tft screen and with the marlin supposedly well configured and I can't get it to work either, I think it may be a problem with the marlin or the board, can you help me, thanks in advance

I plugged it into my SKR 1.4 Turbo. It will not work if you plug it into the TFT screen.