bigtreetech/smart-filament-detection-module #69: SFS v2.0 with Manta M8P - print pause just after wipeline.

krzysiekslimak, 2025-02-09:
I just install SFS v2.0 to my Voron with Manta M8P, I configure it as manual said. Senosors are available in system and looks like they working but every time print start, just after wipe line, the Encoder Sensor throw "runout" and print is paused.

Suraevivan, 2025-05-07:
You must calibrate your detection lenght in Printer.cfg file. Example in User Manual is 2.88 mm just sensor minimum value. You can't use so short distance. In my setup (with direct extruder without bowden tube from sensor to enter the extruder) detection lenght should be more than 50 mm. Try to +10, +20, +30 and more for your printer.

krzysiekslimak, 2025-05-08:
50mm? With this value it will print without filament for quite a long time and You will can't continue print. I leave 2,88mm but add in the print start macro code which turn off SFS2 at the begging and turn it on again after 5 seconds. Then it just work correctly.

Suraevivan, 2025-05-08:
You absolutely right, thats 50 mm only for my setup now, i will make it shorter later after install bowden tube and i think will set 5-10mm. I just mean you must try longer distances and sensor will work correctly.

Suraevivan, 2025-05-09:
Today a tested sensor with bowden tube from sensor to extruder and it works well with detection length 5 mm

[image]